UnipiTechnology/evok #3: Detect and output double tap, triple tap, long press and sustained hold input events

raintonr, 2016-02-01:
As discussed: http://forum.unipi.technology/viewtopic.php?f=14&t=189

> For home automation lighting I'm guessing you'd connect digital inputs to momentary contact light switches.
>
>In such an application it would be very useful to have double tap, triple tap, long press events detected. I'm guessing the best way to do that is in the evok layer and have it spit out different events.
>
>What about sustained holding of an input? Would that be best handled with an 'input on' / 'input off' event pair? Here I'm thinking for controlling dimmable lights or blinds manually rather than setting to predefined levels.

MervColton, 2016-02-21:
I'm planning on having momentary switches for light switches in my house, doing something similar. The useful purpose for me is to be able to turn off ALL the lights in the whole house from the entry/exit door, and from beside the bed. 

I'll be interested in watching this space, and contributing if I can. Just got my unipi this weekend, but forgot to get a power supply :-(

Merv.

MervColton, 2017-03-29:
An update to this: I implemented momentary switches for lights, and it works great. If you do a long-press, over 2 seconds i use that as a signal to turn off all the lights. I implemented this through Python as I could record the start and end time of each press.

PlazzmiK, 2018-02-13:
This would be an awesome feature! I'm currently using Node-RED to implement this because I'm already using it to send everything over MQTT. It would simplify my flow so much.

closed 2026-10-08 by matoushybl