UnipiTechnology/evok #5: Webhook http rest way to get input changes

sworteu, 2016-03-26:
It would be great if one could set a webhook, or another form of feedback without websockets to get input changes.

leogaggl, 2016-11-28:
I would very much second the webhook option...

tknot, 2017-09-15:
Webhook implementation now added in latest Master ( 03ea5ad433271d83d39293c578609d91f2d4111b ).

The implementation may however not work across all servers/interfaces, any feedback is welcome.

wesleydv, 2017-09-29:
The webhook does not provide any data from the triggering event.

I believe the webhook would be much more relevant if the request would carry the data from the triggering event and the values that changed because of it.

tknot, 2017-09-29:
@wesleydv This will be quite easy to add as an optional mode. We are looking to expand the webhook functionality with filtering anyhow (since as it stands the analog outputs generate far too many events), so this will come in the next release.

tknot, 2017-10-11:
@wesleydv Please try the webhook implementation in the current public beta ( d8e4ed7eb04334d5de095e6f084fa8eb1a2ed6fd ) on master. There are two new config options which can be used to customize webhook behavior.

wesleydv, 2017-10-15:
Enabling webhook_complex_events only seems to send an empty post body on sensor updates.
Input changes does not send any request.
webhook_device_mask is set to  ["input", "sensor"]

tknot, 2017-11-02:
This should be fixed in the release candidate 2.0a (and the last beta as well)

closed 2017-11-02 by tknot