UnipiTechnology/evok #15: Neuron set relay output via websocket is not working

martbrugg, 2017-04-06:
Today I tried the latest Master version from Evok with miy Neuron S10x.

But when i tried to set a Digital Output via websocket  the signal did'nt change.
Setting the Output via REST API or HTTP Console works properly.

After installing an older Version of EVOK the websocket was working again.

tomashora, 2017-04-06:
Could you please paste the output when evok runs on hand?

python /opt/evok/evok.py --path1=/etc/evok.conf --path2=/etc/evok-neuron.conf

martbrugg, 2017-04-06:
I uninstalled the old version and reinstalled the new version. And now it is working!!!!
The first installation was fresh raspian jessie OS. Could there be as problem with the setup?

tomashora, 2017-04-06:
What OS are you up now? Usually I am working on latest Raspbian versions and the only issue was to jump to jessie from wheezy but that is already solved.

So closing for now.

closed 2017-04-06 by tomashora