UnipiTechnology/evok #16: Startup error, AttributeError: 'Application' object has no attribute 'handlers'

plReynaerde, 2017-04-17:
After a fresh installation, i got:
root@pascal:~# /opt/evok/evok.py 
Listening on port: 80
UniPi version:1.1
Traceback (most recent call last):
  File "/opt/evok/evok.py", line 545, in <module>
    main()
  File "/opt/evok/evok.py", line 502, in main
    for (srv, urlspecs) in app.handlers:
AttributeError: 'Application' object has no attribute 'handlers'

zusanli, 2017-07-01:
I have the same problem - then I installed the Master and now it works. In the web gui I can't change any settings...

vinjon, 2017-07-03:
What is the Master ?

cervenyj, 2017-07-03:
Same problem here with. I installed master (devel version - see installation instructions) and received above described error. Then I installed stable version and everything (except Settings menu) works fine.

In the past I had similar problem installing first stable version and ony then master (devel). It seems you need to combine both versions to get it work properly ...

All the installations were done on Raspbian Jessie 4.8

Hope this helps,
Jan

mchobby, 2017-07-24:
I have the following error stack error when starting EVOK
`
~ $ sudo /opt/evok/evok.py 
Listening on port: 80
UniPi version:1.1
Traceback (most recent call last):
  File "/opt/evok/evok.py", line 545, in <module>
    main()
  File "/opt/evok/evok.py", line 502, in main
    for (srv, urlspecs) in app.handlers:
AttributeError: 'Application' object has no attribute 'handlers'
`
OS: Raspbian Jessie (2017 07 05) -- installed & updated today
EVOK: Installed with wget by following installation instruction
Running on Python 2.7.9 
Linux raspberrypi 4.9.35-v7+ 

How can I make it working?

mchobby, 2017-07-25:
I have partially solve the problem (cfr settings not working) with the following installation notes about "testing the latest git version".
```
wget https://github.com/UniPiTechnology/evok/archive/master.zip
unzip master.zip
cd evok-master
sudo bash install-evok.sh
```

tknot, 2017-08-09:
This appears to be caused by the latest versions of Tornado removing direct access to the handlers field. The reason Master works is that this part of the code is commented out.

I will update the v.1.0.0 release with a small fix to address this issue, in the meantime it is possible to use the development version instead.

closed 2017-08-09 by tknot

tknot, 2017-08-09:
Fixed legacy release uploaded