UnipiTechnology/evok #18: Service should start even when DS2482 is not present

tomashora, 2017-04-21:
Apr 20 09:17:06 UniPi-Test evok[423]: eprom:: UniPi Neuron S103 version: 1.0 serial: 0x2a
Apr 20 09:17:06 UniPi-Test evok[423]: Using config file /etc/evok-neuron.conf
Apr 20 09:17:06 UniPi-Test evok[423]: HTTP server listening on port: 8088
Apr 20 09:17:06 UniPi-Test evok[423]: Process OwBusDriver-1:
Apr 20 09:17:06 UniPi-Test evok[423]: Traceback (most recent call last):
Apr 20 09:17:06 UniPi-Test evok[423]: File "/usr/lib/python2.7/multiprocessing/process.py", line 258, in _bootstrap
Apr 20 09:17:06 UniPi-Test evok[423]: self.run()
Apr 20 09:17:06 UniPi-Test evok[423]: File "/opt/evok/owclient.py", line 360, in run
Apr 20 09:17:06 UniPi-Test evok[423]: ow.init(self.bus)
Apr 20 09:17:06 UniPi-Test evok[423]: File "/usr/lib/python2.7/dist-packages/ow/__init__.py", line 224, in init
Apr 20 09:17:06 UniPi-Test evok[423]: raise exNoController
Apr 20 09:17:06 UniPi-Test evok[423]: exNoController
Apr 20 09:17:17 UniPi-Test evok[423]: [I 170420 09:17:17 web:1971] 200 GET /rest/all (::1) 4.09ms

phillipsnick, 2017-04-29:
I'm trying to configure DS18B20 temperature sensors on my Neuron N103, after encountering the error `Error in config section OWBUS_1 - No module named pigpio` I followed the instructions for installing pigpio here: http://abyz.co.uk/rpi/pigpio/download.html

But I'm now getting the same error as you have posted above, what actually is the cause of the error?

As the Neuron has the DS2482 chip, I can't see whats missing here? I'm new to 1Wire!

tknot, 2017-09-08:
The overlay for DS2482 is now included in the repository; additionally, the install script has been rewritten to hopefully fix the issue for good. Closing until there are reports of further issues

closed 2017-09-08 by tknot