UnipiTechnology/evok #24: [Errno 111] Connection refused

capitanfindus, 2017-07-09:
Hi,
in Python 2.7, importing from jsonrpclib Server and assigning it to local ip address as per UniPi instruction seems ok.
When trying to set or get relay status, the error in object appears. The other thread in forum did not help.
I can't even open webbrowser with Rpi ip address to get UniPi (Lite) status.

Anyone can help? Thenk you very much

tomashora, 2017-07-10:
It seems evok is not running at all.. https://github.com/UniPiTechnology/evok#debugging

capitanfindus, 2017-07-12:
Thank you Tomas, now it is working. 
It sounds stupid, but I just changed the IP address to the one of the wireless network. Jsonrpclib is working with Python 2.7, as well.
Why was not it working with the eth0 ip address interface?
Do I need to be connected on wifi to use it?

Thank you again

tomashora, 2017-07-13:
We usually test it only when connected to ETH and not using WiFi but the server is listening on 0.0.0.0 so it should accept connection on any interface...

closed 2017-07-19 by capitanfindus