UnipiTechnology/evok #28: evok.conf : port config parameter is ignored

rudyvan, 2017-09-08:
I am using a UniPi version 1.1 on raspian stretch.
I noticed the following issues:

1.  in evok.py, line 2153, port is fixed to 8080 and the port value of evok.conf is not read

2.  in evok.py, line 2240, the soap server port is fixed to 8081 and should be a config parameter

3. further issue is the web version, it is not possible to use  any more

4. when a password is set in the config file, an error is produced when running evok.py:
Traceback (most recent call last):
  File "evok.py", line 2294, in <module>
    main()
  File "evok.py", line 2144, in main
    if pw: userCookieHelper._passwords.append(pw)

it seems that the variable userCookieHelper is not initialised anywhere

phillipsnick, 2017-09-08:
Same issue here with latest master (475cacce2b0e730f87bc42e1d3054b0d98bda579).

It looks like the WebSocket port is hardcoded in various locations inside wsbase.js causing this.

https://github.com/UniPiTechnology/evok/blob/3661ebf02c7036b21abcb301160c709e1c2f3c52/www/evok/js/wsbase.js#L196

tknot, 2017-09-08:
The master branch is now in active development, apologies for any issues it causes. I'll try to answer the questions one by one:
1) This is only temporary for testing, I will commit a fix on Monday
2) Same as 1), however the SOAP interface is not yet finished, which is why it is disabled in the configuration file by default
3) The web version has been moved to NGINX, which is now installed through the installation script. You can access it through the port 88; the port can be changed in /etc/nginx/sites-enabled/evok
4) AFAIK the password has never worked properly, however I will try to get it working ASAP

As for phillipsnick comment, the WebSocket port is now 'hardcoded' in the javascript since the example web interface is now entirely separate, and therefore without the ability to read the EVOK configuration file. The intention is to allow users to connect to EVOK more easily through e.g. external servers. The demonstration website will be updated in due course, for now I would suggest changing the approximately 3 places in the code where it occurs.

It may be that the current master was released prematurely, however the intention was to get a version out which would support all the features of the Neuron line, i.e. LEDs, analogs, etc. Not all of the features which are present in the code are complete as of now, the next major update which will be released next week will include support for Neuron Extensions, after which we will concentrate on making the external interface better.

phillipsnick, 2017-09-08:
Thanks for the update, understand that master is the dev branch just wanted to test the fix for input debounce values on the Neuron line.

tknot, 2017-09-08:
It was really a rather embarrassing issue to persist so long. I'm happy to say that the extensions now appear to be working well on our private (even more development-minded) repository, so we really should get most of the current issues wrapped up within two weeks.

We plan to create a custom image for EVOK as well, since the other Neuron images we currently offer include newer custom drivers with much better performance in certain areas (the serial port chief among them). However the driver has to be compiled for a specific kernel version, and the process of doing so is rather more complicated than for the current version of the TCP Modbus server. In the long term we hope move to a package (i.e. apt-get) system eventually.

Regardless we very much appreciate your interest in Neuron, hopefully you will find the newest additions useful once they are fully fleshed out!

tknot, 2017-09-08:
Closed by accident

rudyvan, 2017-09-09:
changing the port in /etc/nginx/sites-enabled/evok as you mentioned for the web interface does not solve the issue:
If the port number is set to the same as the rest/api in /etc/evok.conf, error 404 is returned and a different port just give back empty web pages, with no relays, inputs or 1w devices.
There need to be more corrected before this works.

Thanks for your effort in updating this software!

tknot, 2017-09-09:
Are you using UniPi 1.1 or Neuron? And have you tried going to port 88, i.e. your.ip.address:88? I have tested the website on a clean raspbian image, so it ought to work.

Though setting it to the same port as the rest API will not work, but I am not entirely sure why would you wish to do so ...

rudyvan, 2017-09-09:
I am using UniPi 1.1 and i have tried on port 88
also i have changed the port to 8180 in nginx but same result
i am using the rest/api, but the web interface is just a quick check if my evok.conf file is ok

in the older (pre neutron) version of evok, it was the same port for the webinterface or for the rest/api but no problem to me if it is different.
all port settings should be defined with parameters in the evok.conf file, trying to figure everything out is rarther time consuming.

tknot, 2017-09-09:
Ahh, we haven't tested the web interface with 1.1; it might be there is an issue somewhere. We'll do some more testing, in the meantime I would suggest reverting to the previous version if it is an issue for you.

rudyvan, 2017-09-09:
would like to migrate to the new version as there is support for the sensor DS2406 that i am using.
but i can wait a few days until the bugs are removed, no issue

tknot, 2017-09-13:
Configuration is now possible via the config file again, as of 666a7b9c241e73ccbfb29ed597c888543edf033b. There will also be a simlink to nginx/evok in the etc directory, however for now the demo website has to be edited directly.

closed 2017-09-13 by tknot