UnipiTechnology/evok #29: evok.conf issue with sensor or 1wdevice

rudyvan, 2017-09-09:
I am using UniPi and i am migrating from the legacy version to the last evok version to use the lastest raspian stretch os.
I carefully updated the new evok.conf version and inserted the [1WDEVICE_xx] and [SENSOR_xx] definitions that i am using and that were working in the legacy version.
When running evok with rest/api, i get an issue that OWSensorDevice does not have an attribute full, and this for all the devices that are defined.
I looked into the config.py source code and it is very different between the legacy and recent version.
There must be an error in the yaml files to return the json string for a OWSensorDevice when is needed to return the rest/api json string.
Can this be corrected?
Thanks in advance

tknot, 2017-09-09:
Could you paste here your evok.conf file? Will look into it

rudyvan, 2017-09-09:
see attached.
Thanks for your help.
KR Rudy

rudyvan, 2017-09-09:
[evok.txt](https://github.com/UniPiTechnology/evok/files/1289970/evok.txt)

rudyvan, 2017-09-12:
This is however a bug, not a question.
Functionality with 1wire DS2408 or registering 1wire devices in evok.conf is broken.
I had to make 2 changes in config.py script to fix it:
class OWSensorDevice():
        ...
	def full(self):
		return self.sensor_dev.full()

and further  ...
        elif devclass == 'SENSOR' or devclass == '1WDEVICE':
               ...
               sensor_dev = owclient....
                # added
	        if typ in ["DS2408","DS2406"]:
			sensor = OWSensorDevice(sensor_dev, dev_id=0)
			Devices.register_device(SENSOR, sensor)
		else: Devices.register_device(SENSOR, sensor_dev)

rudyvan, 2017-09-12:
[config.txt](https://github.com/UniPiTechnology/evok/files/1297024/config.txt)

tknot, 2017-09-12:
Thank you for the fix! It uses older leftover code in a nicely inventive way.

Fact is,  we did not plan to support DS2406 (or 2404/8) officially in the Neuron versions, as the 1Wire devices are autodetected rather than added manually through the config file, which is why I marked this as a question. In fact both me and my colleagues, including one who has written the majority of the code, have been rather confused that the DS2406 works in the latest version at all.

If you look at the line you've commented out, you can see it uses a call to MySensorFabric, which automatically detects and recognises 1Wire devices. The OwSensorDevice et al. classes have only been left in the code by accident, though admittedly so far we can only detect through this method the 1-Wire sensors accompanying the Neuron line. 

However, seeing as we did implement the support at one point (likely by @tomashora) I'm happy to include it, since it should not break anything (I'll probably parameterise out the dev_id into config rather than keeping it fixed, but that's the only thing I can see).

Thank you again

p.s.: The port should now be configurable again through evok.conf.

closed 2017-09-12 by tknot

rudyvan, 2017-09-13:
I noticed the autodetection, and this would be fine, but you still need to "configure" the 1wrelays or 1winputs that are a result of having the DS2406.   Also for backward compatibility it is best for evok.conf to continue to work with older configurations where every 1w device is declared.   
I saw in the webshop that the EMO-1w-8C is not for sale anymore, is that a definitive situation or will it become once again available? 
Thanks!
Rudy

tknot, 2017-09-13:
We currently have no plans to make more the 1Wire extensions, because 1Wire is quite unreliable for control purposes, as you've no doubt found out by now. They were more of a transitory solution.

I would recommend you to look into our new Neuron extensions, though they require a RS485 which UniPi 1.1 does not have.

rudyvan, 2017-09-14:
[image]
little error in the correction, sensor_dev should be used instead of sensor, see the highlighted text

tknot, 2017-09-14:
Ahh right, or it should be sensor both. I don't have a device on hand to
test it with and it was just as last minute change to make sure it's
included. Will fix tomorrow.

rudyvan, 2017-09-15:
just cosmetic, if you redefine the sensor variable, then the shortest version is this:
[image]

tknot, 2017-09-15:
I have tried to avoid significant refactoring for now, but I will change it in the next commit.