UnipiTechnology/evok #30: websocket event for 1-Wire devices hangs after 1 - 2 weeks

g-lom, 2017-09-10:
I have 10 1-Wire Temperatures sensors connected to the Unipi 1.1.
To read the values I use the EVOK (latest version at 23 Aug 2017) websocket event on_message.

This works fine for about 1.5 week. Then no events arrives anymore.
Then I try the following steps:
1. GET value over http API 
==> I get an answer but the time and the value in the response are allways the same
2. I tried to restart the evok process with 'systemctl restart evok'
==> error could not restart the service
3. I tried to reboot (warm restart) the raspberry with 'shutdown -r now'
==> after reboot no Analog Inputs were display on the 'Unipi Control panel' in the browser

Finaly I repluged the 230V Power (cold restart) 

==> Now all Analog Input appeared with correct values

PS.: The same behavior I had already in the EVOK version of january 2017

tknot, 2017-09-11:
This appears to be a hardware issue with interference on the UniPi 1.1 1-Wire/I2C interface chip unrelated to EVOK, partly based on the fact that the Analog inputs (which are also on I2C) exhibit this behaviour, as well as the warm restart (which from a software POV is a complete restart) does not help.  We've seen a similar problem before with other customers. I would suggest making sure that there is good grounding and isolation, and perhaps try to rethink the cabling to avoid potentially interference hazards.

tknot, 2017-09-11:
Closed for now

closed 2017-09-11 by tknot