UnipiTechnology/evok #31: Release v1.0.2 Breaks some API functionality and web interface

joearkay, 2017-09-12:
Device: Neuron S103
OS: Tried on both raspbian Jessie and Stretch

Managed to access some endpoints via GET with mixed results. No response from any POSTs. 404 when trying to access webpage. 

Confirmed that it is an issue with 1.0.2 by installing 1.0.1 successfully.

tknot, 2017-09-12:
Will look into it; however the only change between 1.0.1 and 1.0.2 is, well, this: 4bbd96c7a1a1a93ec6c08cd1e0069d5ba052c454

Specifically the change was made to make it possible to disable the evok service, as on newer versions of systemd the alias would fail to remove.

tknot, 2017-09-12:
The only possibility I can see is that EVOK perhaps fails to run, given the only change between the two versions.

Would it be possible to run it manually (python /opt/evok.py) or post the output given by "systemctl start evok"?

tknot, 2017-09-14:
Looking into this further I've realized that the issue here is more that legacy releases do not in fact include support for Neuron, unlike the current Master. This is true of both **1.0.1 _and_ 1.0.2**, which is what misled me. 

It appears that you must have inadvertently downloaded Master instead of 1.0.1, as 1.0.1 (and 1.0.2) is based on codebase dating before the commercial release of Neuron considerable amount of time back, and therefore could not possibly function with it.

The fact that Neuron is supported only in the Master branch is noted clearly in the readme file which opens when visiting the directory.

tknot, 2017-09-14:
Closed as "not an issue"

closed 2017-09-14 by tknot