UnipiTechnology/evok #32: Cannot get Evok to work on UniPi Neuron S103

joearkay, 2017-09-15:
Hi, 

have previously setup 10+ Neuron S103s with Evok, given a few months ago. 

However recently I cannot access the web interface when installing. I get a '404'. I've tried changing port numbers etc but I get nothing. Always 404. 

Installing PIGPIO first: 

```
wget abyz.co.uk/rpi/pigpio/pigpio.zip
unzip pigpio.zip
cd PIGPIO
make
sudo make install
```

Installing Evok like so: 

```
wget https://github.com/UniPiTechnology/evok/archive/master.zip
unzip master.zip
cd evok-master
sudo bash install-evok.sh (you miss the sudo from your guide)
```

Select '3' option for Neuron and then reboot when prompted. 

Once rebooted:

`sudo service evok status` tells me:

```
● evok.service - Evok Modbus/Websocket/Rpc Server
   Loaded: loaded (/etc/systemd/system/evok.service; enabled)
   Active: active (running) since Fri 2017-09-15 11:30:52 UTC; 3min 22s ago
 Main PID: 1367 (python)
   CGroup: /system.slice/evok.service
           ├─1367 /usr/bin/python /opt/evok/evok.py
           └─1374 /usr/bin/python /opt/evok/evok.py

Sep 15 11:30:52 raspberrypi systemd[1]: Started Evok Modbus/Websocket/Rpc Server.
```

So I run `sudo service evok stop`, then modify the conf file to a free port, say `21555`. 

I then run `sudo service evok start` and enter `192.168.1.172:21555` (ip of my UniPi) on my computer's chrome browser, and get `404: Not Found`. 

Running raspbian on this UniPi. 

I had written myself a full guide a month or so ago to follow the steps, and it would install perfectly every time - what's changed? 

Thanks

joearkay, 2017-09-15:
```
pi@raspberrypi:~ $ systemctl stop evok
Failed to stop evok.service: Access denied
pi@raspberrypi:~ $ sudo systemctl stop evok
pi@raspberrypi:~ $ sudo python /opt/evok/evok.py
[I 170915 11:52:46 config:68] eprom: UniPi Neuron S103 version: 1.0 serial: 0xa4
[I 170915 11:52:46 evok:2181] Starting using config file /etc/evok.conf
[I 170915 11:52:46 config:84] YAML Definition loaded: L40x.yaml, type: 2, definition count 0
[I 170915 11:52:46 config:84] YAML Definition loaded: M40x.yaml, type: 2, definition count 1
[I 170915 11:52:46 config:84] YAML Definition loaded: S10x.yaml, type: 4, definition count 2
[I 170915 11:52:47 config:84] YAML Definition loaded: L51x.yaml, type: 2, definition count 3
[I 170915 11:52:47 config:84] YAML Definition loaded: xS50.yaml, type: 2, definition count 4
[I 170915 11:52:47 config:84] YAML Definition loaded: M20x.yaml, type: 2, definition count 5
[I 170915 11:52:47 config:84] YAML Definition loaded: M30x.yaml, type: 2, definition count 6
[I 170915 11:52:47 config:84] YAML Definition loaded: L20x.yaml, type: 2, definition count 7
[I 170915 11:52:47 config:84] YAML Definition loaded: L30x.yaml, type: 2, definition count 8
[I 170915 11:52:47 config:84] YAML Definition loaded: xS30.yaml, type: 2, definition count 9
[I 170915 11:52:47 config:84] YAML Definition loaded: M10x.yaml, type: 4, definition count 10
[I 170915 11:52:47 config:84] YAML Definition loaded: DOMAT MMIO.yaml, type: 2, definition count 11
[I 170915 11:52:47 config:84] YAML Definition loaded: M50x.yaml, type: 2, definition count 12
[I 170915 11:52:48 config:84] YAML Definition loaded: L50x.yaml, type: 2, definition count 13
[I 170915 11:52:48 config:84] YAML Definition loaded: xS10.yaml, type: 2, definition count 14
[I 170915 11:52:48 config:84] YAML Definition loaded: xS40.yaml, type: 2, definition count 15
[I 170915 11:52:48 evok:2262] HTTP server listening on port: 21555
CLIENT STARTED
READING BOARDS SPI
[1280, 1028, 785, 16, 16, 195, 0, 5, 5000, 12184]
<class 'pymodbus.register_read_message.ReadInputRegistersResponse'>
21
32
1
[D 170915 11:52:48 owclient:370] Entering 1wire loop
NOBOARD
NOBOARD
[D 170915 11:52:48 owclient:299] New sensor DS18B20 - 288222AA08000076
ERGO 1
[0, 0, 0, 3, 9, 11306, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
[0, 0, 0, 3, 9, 11306, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
[<neuron.AnalogInput instance at 0x75e50468>]
ERGO 1
[0, 0, 0, 4, 8, 11306, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
[0, 0, 0, 4, 8, 11306, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
[<neuron.AnalogInput instance at 0x75e50468>]
ERGO 1
[0, 0, 0, 4, 8, 11306, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
[0, 0, 0, 4, 8, 11306, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
[]
ERGO 1
[0, 0, 0, 4, 8, 11306, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
[0, 0, 0, 4, 8, 11306, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
[]
ERGO 1
[0, 0, 0, 4, 9, 11306, 8, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
[0, 0, 0, 4, 9, 11306, 8, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
[<neuron.Watchdog object at 0x75fa06b0>]
ERGO 1
[0, 0, 0, 4, 8, 11306, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
[0, 0, 0, 4, 8, 11306, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
[<neuron.Watchdog object at 0x75fa06b0>]
ERGO 1
[0, 0, 0, 4, 9, 11306, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
[0, 0, 0, 4, 9, 11306, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
[]
ERGO 1
[0, 0, 0, 3, 8, 11306, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
[0, 0, 0, 3, 8, 11306, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
[<neuron.AnalogInput instance at 0x75e50468>]
ERGO 1
[0, 0, 0, 4, 8, 11306, 8, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
[0, 0, 0, 4, 8, 11306, 8, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
[... 102 lines cut ...]
[]
ERGO 1
[0, 0, 0, 3, 8, 11306, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
[0, 0, 0, 3, 8, 11306, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
[<neuron.AnalogInput instance at 0x75e50468>]
ERGO 1
[0, 0, 0, 3, 8, 11306, 8, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
[0, 0, 0, 3, 8, 11306, 8, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
[<neuron.Watchdog object at 0x75fa06b0>]
ERGO 1
[0, 0, 0, 4, 8, 11306, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
[0, 0, 0, 4, 8, 11306, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
[<neuron.AnalogInput instance at 0x75e50468>, <neuron.Watchdog object at 0x75fa06b0>]
ERGO 1
[0, 0, 0, 4, 8, 11306, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
[0, 0, 0, 4, 8, 11306, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
[]
ERGO 1
[0, 0, 0, 3, 8, 11306, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
[0, 0, 0, 3, 8, 11306, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
[<neuron.AnalogInput instance at 0x75e50468>]
^C[I 170915 11:53:08 evok:2325] Shutting down
```

There's a bit of logging. As you can see, it 404s in the logs also, confirming it's not an issue client-side, at the web-browser

tomashora, 2017-09-15:
Hello @joearkay, in the last commits the actual web server interface has been removed from evok and is now handled by nginx so evok is now really only a API separated from the web interface. So the port you define in evok.conf is the port of the API but not the port of the Web interface.

By default, the evok API port is 8088 - https://github.com/UniPiTechnology/evok/blob/master/etc/evok-neuron.conf#L15

and is used by the web application -
 https://github.com/UniPiTechnology/evok/blob/master/www/evok/js/wsbase.js#L6

which is handled by the nginx server on port 88 (this is the port to open in your browser to see the web interface)- https://github.com/UniPiTechnology/evok/blob/master/www/evok/js/wsbase.js#L6

This will be described during the installation process and you will be able to choose which ports will be used by Evok and the web service.

rudyvan, 2017-09-15:
i also noticed that when nginx is installed as a result of installing evok, that port 80 is now claimed by nginx as a default server port.
This may clash with applications where port 80 is used by other applications.
This is new aspect not previously there when installing older versions of evok. 
so you need to edit /etc/nginx/sites-enabled/default and change the port to something harmless.
I did not research how to disable the default nginx webserver after installation.

joearkay, 2017-09-15:
Hi @tomashora - Thanks for the update. Is there a place where this is docced? Any release notes or updates to the readme.md? I was not aware of these changes. Thanks 👍

tomashora, 2017-09-15:
@joearkay not yet..But definitely have to be :)

joearkay, 2017-09-15:
@tomashora - Yes please, should be updated with every release! This cost me nearly 2 days! 

As a side note, the new web app doesn't show any of my S103's I/O...likewise the conf file isn't populated with my one-wire/digital/analogue I/O. Has the auto-detect been disabled? 

Aside to this, I'm constantly getting a 'connection is closed' dialogue box appear when viewing the web app, happens every 5 seconds or so. 

Thanks

joearkay, 2017-09-15:
Also, evok seems to start on boot now, when it never did before???

joearkay, 2017-09-15:
Seems to have randomly autodiscovered now...

joearkay, 2017-09-15:
Sorry @tknot / @tomashora - Is there a way I can force the autodiscovery of I/O? I've been waiting around 15 minutes on another UniPi and the web interface just won't populate.

tknot, 2017-09-15:
The autodiscovery for non-1Wire devices happens only when EVOK starts. This is likely a port issue; particularly given the warnings you get, as they signify that EVOK cannot connect to the correct port.

If you download the current master ( 86c2bac7ad6d12d906aceccd992bd2a75334842a ) you should be able to access everything via port 80 again (if you wish to use a different port instead, it can be changed in /etc/nginx/sites-enabled/evok).

We are trying to make everything as simple as possible, but sometimes it is hard for us to see how difficult the configuration may be for our users.

rudyvan, 2017-09-15:
Just a question.
Is there any plan to migrate from tornado and nginx just to nginx?
Having 2 webservers mixed in one script with different config files is just one too much.

joearkay, 2017-09-16:
@tknot Thank you. Do you know the commit hash of the version I would have downloaded a couple of months ago? The version that used the previous web interface? I'd like to download this and continue to use it, until these issues are rectified. Thanks 👍

tknot, 2017-09-16:
@joearkay You are welcome to use the version at ( 97c9d239c23691d6780d578359dca04b2c6e74c5 ), but do note that the interface there has some issues of it's own.

@rudyvan It's quite common to use 2 webservers, one to serve static content and another to serve dynamic one. In fact it's considerably faster than using a single one, particularly given that Tornado is not parallelised in it's main loop, while the raspberry Pi has multiple cores.

tknot, 2017-09-16:
As an additional explanation, nginx is not capable of, or optimised for, serving dynamic content, such as an API is. In addition the web interface is quite limited in it's current (and past) revision and is not intended to be the primary means of accessing EVOK.

joearkay, 2017-09-18:
Not sure why this was closed, as my issue hasn't really been rectified. As we use UniPi commercially I might send you an email @tomashora

tomashora, 2017-09-18:
@joearkay since https://github.com/UniPiTechnology/evok/commit/86c2bac7ad6d12d906aceccd992bd2a75334842a the Nginx is proxying API requests to Evok and servers only the static pages. So the behaviour is the same as before but you do not change the website port in evok.conf but in the Nginx site configuration.

So right now, default port for both WEB and API is 80 (just as before): https://github.com/UniPiTechnology/evok/blob/master/etc/nginx/sites-enabled/evok#L3

Besides that Evok is listening on 8088: https://github.com/UniPiTechnology/evok/blob/master/etc/evok-neuron.conf#L15

This will be later described in the readme. 

But sure, contact us and send us some information about your application so we can understand it talk about specific issues.

As I understand the issue was caused by the undocumented change which is now described here.

rudyvan, 2017-09-19:
I don't want to be annoying, but i think that issue is not solved.

I am working with UniPi 1.1 and evok and i used to allocate port 8181 for both api as webserver.

if in /etc/evok/evok.conf the port is the same as in /etc/nginx/site-enabled/evok, then nginx does not restart and i get a static webpage without content (see picture).

If the port numbers are different, then if get an empty web page and connection aborted.

Whatever i do, i cannot seem to get the webserver working, only the api.

[image]
i am using the last release.

tknot, 2017-09-19:
@rudyvan 
Definitely seems like we need to make the configuration easier. We assumed it's not actually all that difficult, but clearly that's not the case. I already have a changed installation script which allows for specifying the port during installation, and we might think of making some sort of tutorial.

Regarding your issue - NGINX must be running as you would not be able to see even the token interface otherwise, however it looks like it cannot connect to EVOK. Could you please post your sites-enabled/evok config?

rudyvan, 2017-09-20:
I don't mind changing 2 different conf files, i just want to get it work.
As per your request, see the attached /etc/nginx/sites-enabled/evok and the /etc/evok.conf file.
both have a different port addr, only the api on address 8181 is working, the webserver on 8180 is not.
I think there must be another port for the tornado server to work, but i don't have time to dig into the source code.  Thanks for the help.

[image]

rudyvan, 2017-09-20:
seems the evok file got lost, see here

[evok.conf.txt](https://github.com/UniPiTechnology/evok/files/1318082/evok.conf.txt)

tknot, 2017-09-21:
Ahh you have to change the "proxy_pass http://localhost::8080" lines to "proxy_pass http://localhost::8181". If you download the current master (or in the last two days) the installation script will ask which two ports you wish to use, for the webserver and the API, and will set everything up appropriately.

closed 2017-09-21 by tknot

rudyvan, 2017-09-21:
ok, now i understand, in /etc/nginx/sites-enabled/evok the listen port is the external one and the proxy_pass reads from the rest api and therefore must be the internal port.
Mistery solved.     The changed install script will help to avoid this.   Thanks.