UnipiTechnology/evok #33: Move web app configuration to a separate file

tomashora, 2017-09-15:
Allow clear definition of the connection parameters to the Evok API from web app since both applications can now run on different hosts

tknot, 2017-09-15:
The API has now been moved through redirection to port 80, which is now shared with EVOK by default.

For anyone using our Mervis image, the port will be 88 instead, to avoid conflicts with the HMI service.

The website configuration interface remains unfinished for now.

tknot, 2017-10-09:
The installation script has been updated to allow specifying the port to use; furthermore in the next release a symlink from /etc/nginx/sites-enabled/evok has been made to /etc/evok-web.conf. Some more explanation it still needed, but for now the issue is mostly resolved

tknot, 2017-10-11:
There is now a symlink to the file in /etc, as of ( d8e4ed7eb04334d5de095e6f084fa8eb1a2ed6fd )

closed 2017-10-11 by tknot