UnipiTechnology/evok #34: HTTP index.html not showing up, gives 404 instead

eLvErDe, 2017-09-15:
Hi,

With today's GIT master I cannot see the usual webinterface. All I get is 404, despite the other routes seem to work correctly.

Here's app.handlers for tornado webapp:

```
[(<_sre.SRE_Pattern object at 0x76367a20>,
  [URLSpec('/auth/login/$', <class '__main__.LoginHandler'>, kwargs={}, name=None),
   URLSpec('/auth/logout/$', <class '__main__.LogoutHandler'>, kwargs={}, name=None),
   URLSpec('/rpc$', <class 'rpc_handler.Handler'>, kwargs={}, name=None),
   URLSpec('/config$', <class '__main__.ConfigHandler'>, kwargs={}, name=None),
   URLSpec('/config/cmd$', <class '__main__.RemoteCMDHandler'>, kwargs={}, name=None),
   URLSpec('/json$', <class '__main__.JSONHandler'>, kwargs={}, name=None),
   URLSpec('/rest/all/?$', <class '__main__.LoadAllHandler'>, kwargs={}, name=None),
   URLSpec('/rest/([^/]+)/([^/]+)/?([^/]+)?/?$', <class '__main__.LegacyRestHandler'>, kwargs={}, name=None),
   URLSpec('/ws$', <class '__main__.WsHandler'>, kwargs={}, name=None)])]
```

Can't see anything related ton /, index, static or what ever...

Thanks in advance,

Adam.

tomashora, 2017-09-15:
Similar to https://github.com/UniPiTechnology/evok/issues/32 ?

eLvErDe, 2017-09-15:
Oh, indeed...

Installed a webserver and I can now see the interface. However, there's no in/out available but I see hundreds of errors in evok.log :(

eLvErDe, 2017-09-15:
Seeing stuff like this:

```
[E 170915 14:33:23 config:359] Error in config section MCP_1 - 'I2C write failed'
    Traceback (most recent call last):
      File "/opt/evok/config.py", line 242, in create_devices
        mcp = unipig.UnipiMcp(bus, circuit, address=address, dev_id=0)
      File "/opt/evok/unipig.py", line 85, in __init__
        i2cbus.i2c_write_byte_data(self.i2c, MCP23008_IODIR, 0x00)  # all output !
      File "/usr/lib/python2.7/dist-packages/pigpio.py", line 2684, in i2c_write_byte_data
        self.sl, _PI_CMD_I2CWB, handle, reg, 4, extents))
      File "/usr/lib/python2.7/dist-packages/pigpio.py", line 970, in _u2i
        raise error(error_text(v))
    error: 'I2C write failed'
[E 170915 14:33:23 config:359] Error in config section RELAY_1 - Invalid device circuit number 1
    Traceback (most recent call last):
      File "/opt/evok/config.py", line 247, in create_devices
        mcp = Devices.by_int(MCP, mcp)
      File "/opt/evok/devices.py", line 32, in by_int
        raise Exception('Invalid device circuit number %s' % str(circuit))
    Exception: Invalid device circuit number 1
[E 170915 14:33:23 config:359] Error in config section RELAY_2 - Invalid device circuit number 1
    Traceback (most recent call last):
      File "/opt/evok/config.py", line 247, in create_devices
        mcp = Devices.by_int(MCP, mcp)
      File "/opt/evok/devices.py", line 32, in by_int
        raise Exception('Invalid device circuit number %s' % str(circuit))
    Exception: Invalid device circuit number 1
```

It's my first time on Neuron, I used to work with UniPi and it's a lot more out-of-the-box ;)

tknot, 2017-09-15:
It seems you are using a configuration file from UniPi 1.1, based on the errors you get. 

Could you try re-running the installation script for Master (the instruction are in readme.md) and selecting Neuron (i.e. 3) when prompted?

tknot, 2017-09-15:
If you reinstall evok from scratch with the current master, you should be able to access it through port 80 (i.e. the default HTTP/web port) again.

I am sorry for any issues caused by the continuing development, it's inevitable that as things get added certain features/information may be slightly at odds with how they were before.

I have also changed the install script to delete the default Nginx project running on port 80.

closed 2017-09-16 by tknot