UnipiTechnology/evok #35: Where are unipi2_eprom / unipi_rtc kernel modules sources ?

eLvErDe, 2017-09-20:
Trying to install Neuron from scratch on Raspbian, but I miss these modules...

Thanks in advance, Adam.

tknot, 2017-09-21:
These overlays are only for UniPi 1.1 (slightly cofusing naming scheme perhaps) rather than Neuron, so you do not need them. 

They also do not have kernel modules, unlike the neuronspi they are purely hardware configuration overlays.

tknot, 2017-09-21:
Looking into it more there were modules of those names which have been used in earlier revisions (and remain in modprobe.d). Regardless, they are not necessary, the overlays in /boot/overlays should be sufficient.

eLvErDe, 2017-09-21:
Make sense but it does not work for me.

Here is my config.txt:
```
[pi3]
kernel=vmlinuz-4.9.0-2-rpi2
initramfs initrd.img-4.9.0-2-rpi2 followkernel
device_tree_param=i2c1=on
dtoverlay=i2c-rtc,mcp7941x
dtoverlay=unipiee
dtoverlay=neuronee
dtoverlay=neuron-spi
```

Overlays in /boot
```
-rwxr-xr-x 1 root root 2.6K Sep 20 09:30 /boot/i2c-rtc.dtbo
-rwxr-xr-x 1 root root  675 Sep 20 09:29 /boot/neuronee.dtbo
-rwxr-xr-x 1 root root 1.7K Sep 20 09:05 /boot/neuron-spi.dtbo
-rwxr-xr-x 1 root root  674 Sep 20 09:30 /boot/unipiee.dtbo
```
But when I start Evok, it fails to identify Neuron version correctly:

Method `def read_eprom_config():` try to do the following:
```
        try:
                with open('/sys/class/i2c-dev/i2c-1/device/1-0057/eeprom','r') as f:
                        bytes=f.read(128)
                        if bytes[96:98] == '\xfa\x55':
                                globals['version2'] = "%d.%d" % (ord(bytes[99]), ord(bytes[98]))
                                globals['model'] = "%s" % (bytes[106:110],)
                                globals['serial'] = struct.unpack('i', bytes[100:104])[0]
                                logger.info("eprom: UniPi Neuron %s version: %s serial: 0x%x", globals["model"], globals['version2'],globals["serial"])
        except Exception, E:
                pass
```
However this file does not exist:
`ls: cannot access /sys/class/i2c-dev/i2c-1/device/1-0057/eeprom: No such file or directory`

But I do have something on i2c 0x57:
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- 18 -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- 57 -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- UU 
70: -- -- -- -- -- -- -- --  
```

Would you be kind enough to look into this ? I running out of ideas.

Thanks, Adam.

tknot, 2017-09-21:
Do you have the overlays in the /boot/overlays/ or /boot/ ? They should be in /boot/overlays/.

closed 2017-09-21 by tknot

eLvErDe, 2017-09-21:
Good catch indeed, that was it... thanks.