UnipiTechnology/evok #36: Websocket cmd:all

flettieri, 2017-09-23:
Hi all
the websocket cmd:all is affected by bug. Every time I run it the following exception is raised:
`unsupported operand type(s) for +=: 'dict' and 'list'`
At line 148 
`result = {}`
is initialized as dict but later in line 152 : 
`result += map(lambda dev: dev.full(), Devices.by_int(dev))`
there is a concatenation between dict (result) and list (the result of the fuction dev.full() )
A quick workaround to solve the issue is initialize result as list
`result=[]`

tknot, 2017-09-24:
Confirmed in the latest master; issue same as https://forum.unipi.technology/topic/431/problem-using-cmd-all-using-websockets - appears to have resurfaced?

tknot, 2017-09-24:
Thank you for the report! I will fix this within the next public master commit.

tknot, 2017-10-11:
This should now be fixed again, as of ( d8e4ed7eb04334d5de095e6f084fa8eb1a2ed6fd )

tknot, 2017-11-02:
Marked as fixed for now

closed 2017-11-02 by tknot