UnipiTechnology/evok #37: DS2413 is not supported

tomashora, 2017-09-25:
https://forum.unipi.technology/topic/458/unipi-is-not-detecting-ds2413

tknot, 2017-09-25:
DS2413 is very similar to DS2406/8 which we already support, and should be easy to implement

tknot, 2017-09-26:
The support for DS2413 will be available in the next release

tknot, 2017-10-11:
The support for DS2413 should now be included, as of ( d8e4ed7eb04334d5de095e6f084fa8eb1a2ed6fd )

seecosmea, 2017-10-16:
I followed your instructions and downloaded & installed the new EVOK release (d8e4ed7) - [NEURON] DS2413 is not supported #37.

All communication to my UniPi board (1.1) is lost after instalation of the new Evok software. 

REST API, HTTP GET & Websocket are not more working. 
http://127.0.0.1:88 showing Unipi side without any status - no communication as well 

Pls see below:

i@raspberrypi:~ $ sudo python /opt/evok/evok.py
[I 171016 08:42:00 evok:2275] Starting using config file /etc/evok.conf
[I 171016 08:42:00 config:84] YAML Definition loaded: xS50.yaml, type: 3, definition count 0
[I 171016 08:42:00 config:84] YAML Definition loaded: L51x.yaml, type: 3, definition count 1
[I 171016 08:42:00 config:84] YAML Definition loaded: CUSTOM MODBUS DEVICE.yaml, type: 2, definition count 2
[I 171016 08:42:00 config:84] YAML Definition loaded: M20x.yaml, type: 3, definition count 3
[I 171016 08:42:01 config:84] YAML Definition loaded: M30x.yaml, type: 3, definition count 4
[I 171016 08:42:01 config:84] YAML Definition loaded: M10x.yaml, type: 3, definition count 5
[I 171016 08:42:01 config:84] YAML Definition loaded: L50x.yaml, type: 3, definition count 6
[I 171016 08:42:01 config:84] YAML Definition loaded: xS40.yaml, type: 3, definition count 7
[I 171016 08:42:01 config:84] YAML Definition loaded: L20x.yaml, type: 3, definition count 8
[I 171016 08:42:01 config:84] YAML Definition loaded: M50x.yaml, type: 3, definition count 9
[I 171016 08:42:02 config:84] YAML Definition loaded: M40x.yaml, type: 3, definition count 10
[I 171016 08:42:02 config:84] YAML Definition loaded: L30x.yaml, type: 3, definition count 11
[I 171016 08:42:02 config:84] YAML Definition loaded: L40x.yaml, type: 3, definition count 12
[I 171016 08:42:02 config:84] YAML Definition loaded: xS30.yaml, type: 3, definition count 13
[I 171016 08:42:02 config:84] YAML Definition loaded: xS10.yaml, type: 3, definition count 14
[I 171016 08:42:02 config:84] YAML Definition loaded: S10x.yaml, type: 3, definition count 15
Traceback (most recent call last):
  File "/opt/evok/evok.py", line 2436, in <module>
    main()
  File "/opt/evok/evok.py", line 2279, in main
    alias_dict = config.HWDict('/var/evok')
  File "/opt/evok/config.py", line 79, in __init__
    for filen in os.listdir(d_path):
OSError: [Errno 2] No such file or directory: '/var/evok'

#!!! Do not use '#' for comments !!!

[MAIN]
config_version = 2.2
use_experimental_api = False                    ; Currently in Alpha - known issues
regenerate_api_docs = False                             ; Active only with the experimental API setting
log_level = ERROR                                       ; One of INFO, DEBUG, WARNING, ERROR, CRITICAL
log_file = /var/log/evok.log                    ; Log file to use; will be cleared on boot
;password = test
;rpcpassword = test
secret = ut5kB3hhf6VmZCujXGQ5ZHb1EAfiXHcy
port = 8080                                                             ; Internal API port - only change if you are certain you know what you are doing; !!!For our web interface the port should be changed in /etc/evok-nginx.conf instead$
;modbus_port = 0
;modbus_address = *
webhook_enabled = False                                 ; Enables webhook notification - see e.g. https://sendgrid.com/blog/whats-webhook/
webhook_address = http://127.0.0.1:80   ; Put your server endpoint address here (e.g. http://123.123.123.123:/wh )
webhook_device_mask = input:relay:wd    ; List of device types to notify on (separated by ':') - do not add AI unless you want to be swamped with messages!
webhook_complex_events = False                  ; EVOK will send POST requests with the same data as WebSocket, rather than an empty GET request
[OWBUS_1]
owbus = --i2c=/dev/i2c-1:ALL     ; scanned bus (--i2c=/dev/i2c-1:ALL or localhost:2122 or 'u' for USB dongle)
interval = 3          ; [s] default sensor reading
scan_interval = 300    ; [s] how often is made searching

;[SENSOR_1]          ; onewire thermometer
;bus = 1                     ; link to OWBUS_X
;address = 28527B930500001F
;type = DS18B20
;interval = 1

[I2CBUS_1]
busid = 1               ;[integer] id of bus /dev/i2c-X

;EPROM device will not be used anymore, eprom should be handled by kernel module
;[EPROM_1]
;i2cbus = 1
;address = 0x50
;size = 256

I appreciate your assistance

tknot, 2017-10-16:
Ahh it looks like the installation script hasn't been properly updated for UniPi 1.1 yet.

You can fix the issue by calling "mkdir /var/evok" (as root).

tknot, 2017-10-16:
This should now be fixed in ( 45b9d3ef6df280843229a6473b34e8762d8fd0a3 )

seecosmea, 2017-10-16:
REST API, HTTP GET are working now, Websocket following problem:

[Errno -2] Name or service not known

Traceback (most recent call last):
  File "/home/pi/Desktop/WS/new.py", line 21, in <module>
    ws.run_forever()
  File "/usr/lib/python2.7/dist-packages/websocket/_app.py", line 179, in run_forever
    self.sock.close()
  File "/usr/lib/python2.7/dist-packages/websocket/_core.py", line 819, in close
    self._closeInternal()
  File "/usr/lib/python2.7/dist-packages/websocket/_core.py", line 822, in _closeInternal
    self.sock.close()
AttributeError: 'NoneType' object has no attribute 'close'

was using following code:

import websocket
import json

url = "ws://your.127.0.0.1/ws"

def on_message(ws, message):
    obj = json.loads(message)
    dev = obj['dev']
    circuit = obj['circuit']
    value = obj['value']
    print message

def on_error(ws, error):
    print error

def on_close(ws):
    print "Connection closed"

#receiving messages
ws = websocket.WebSocketApp(url, on_message = on_message, on_error = on_error, on_close = on_close)
ws.run_forever()

#sending messages
ws = websocket.WebSocket()
ws.connect(url)
ws.send('{"cmd":"set","dev":"relay","circuit":"3","value":"1"}')
ws.close()

Furthermore, DS2413 is not visible / detected via  http://127.0.0.1:88 and / or other methodes. DS2413

seecosmea, 2017-10-16:
The webpage http://127.0.0.1:88 is showing under: Inputs / 1 Wire Sensors / Devices my DS18B20 temperatur sonsors only but not the DS2413 Adafruit 1-Wire GPIO

seecosmea, 2017-10-16:
The WebSocket is working now; sorry for inconvinicence caused but the issure with the DS2413 Adafruit 1-Wire GPIO is not yet solved.

Greetings from Dubai; I have to work now; I will be back in about 1 week, tks in advance for assistance....

seecosmea, 2017-10-24:
The new EVOK release is detecting DS2413 now, I receive the OW information string. Tks for all your work here ! Greetings from India - Cochin:)

tknot, 2017-10-24:
Considered fixed unless further issues arise

closed 2017-10-24 by tknot