UnipiTechnology/evok #38: [NEURON] Can't start evok after fresh install on UniPian-Neuron-OS-2017-08-30

wesleydv, 2017-09-25:
I installed evok on a freshly installed UniPian-Neuron-OS-2017-08-30 by first running the command `mount -o remount,rw /`.
Then following the instructions in the readme and rebooting.
But the evok service does not start.

When manually starting the python program I get:
```
root@unipi:~# python /opt/evok/evok.py
Traceback (most recent call last):
  File "/opt/evok/evok.py", line 35, in <module>
    import config
  File "/opt/evok/config.py", line 7, in <module>
    import neuron
  File "/opt/evok/neuron.py", line 25, in <module>
    import modbusclient_rs485
  File "/opt/evok/modbusclient_rs485.py", line 52, in <module>
    from pymodbus.compat import byte2int
ImportError: No module named compat
```

tknot, 2017-09-26:
It looks like there are some issues with the open-source image and evok. It might be better to use Stretch for now, I will update the readme file accordingly. 

If you wish to run it regardless. you can use the commands below, but they will need to be used on every reboot. I would recommend using standard Stretch instead.

`mount -o rw,remount /
pip install pymodbus --upgrade
systemctl enable evok
systemctl start nginx
systemctl start evok`

wesleydv, 2017-09-26:
Thanks for the info.

I've switched to stretch after posting this bug and everything is working fine.

tknot, 2017-11-02:
Marked closed for now, new image will be made once the 2.0 release is finalized, which should be in 2 weeks time.

closed 2017-11-02 by tknot