UnipiTechnology/evok #39: [NEURON] Direct Switch access

wesleydv, 2017-09-29:
Is there a way I can access the Direct Switch settings via EVOK?

tknot, 2017-09-29:
This will be coming in the next beta release. EVOK was never completely finished for Neuron before the Neuron hardware was released, which is why it is still in beta after several months. 

I will try to get the next beta released early next week, with the update hopefully being the last focusing on the backend.

tknot, 2017-10-11:
This should now be solved in the latest beta ( d8e4ed7eb04334d5de095e6f084fa8eb1a2ed6fd ). Direct switch can be accessed by setting the 'mode' variable of an input to the string 'DirectSwitch' (available modes are listed in the output returned from a GET to '/rest/all'). 

Direct switch behavior can then be set via the "ds_mode" variable, which appears after switching the input into DirectSwitch mode.

tknot, 2017-11-02:
Marked as fixed now, see the api-docs in Master readme for more information

closed 2017-11-02 by tknot