UnipiTechnology/evok #40: [NEURON] Error while connection extension module

wesleydv, 2017-09-29:
I'm trying to connect a xS40 to my neuron. 
The config is set in /etc/evok.conf and communication between two seems to be ok, TX and RX leds are blinking on both modules, the run led is solid on the xS40 module.

However the inputs and outputs to not appear in the API.

When I start the service manually I can see the following error during startup:
```
[E 170929 20:45:35 neuron:242] list index out of range
    Traceback (most recent call last):
      File "/opt/evok/neuron.py", line 231, in readboards
        board = UartBoard(self.Config, self.circuit, self.modbus_address, self, versions.registers, dev_id=self.dev_id)
      File "/opt/evok/neuron.py", line 304, in __init__
        self.sw = versions[0]
    IndexError: list index out of range
[1283, 4110, 0, 2064, 784, 5803, 0, 4, 5000, 1521]
<class 'pymodbus.register_read_message.ReadInputRegistersResponse'>
```

tknot, 2017-10-02:
It looks like the modules cannot communicate together. It's not immediately obvious to me what is going wrong. Do the issues still persist?

It may be the hardware definition file is wrong, does it contain the "first_board_conf_count" key?

In any case there should be a number of improvements to the rs485 functionality in the next release sometime this week, which will hopefully deal with the issue.

tknot, 2017-10-02:
You can also try the current master commit as a workaround for now.

wesleydv, 2017-10-02:
Since my device name is xS40 I believe I'm using this file: https://github.com/UniPiTechnology/evok/blob/master/etc/hw_definitions/xS40.yaml as hardware definition file, and it does not contain the "first_board_conf_count" key.

When I try the latest commit I get no errors on startup however when calling /rest/all I get a 500 and these error's appear in the log:

```
[E 171002 09:50:29 web:1590] Uncaught exception GET /rest/all (127.0.0.1)
    HTTPServerRequest(protocol='http', host='127.0.0.1:8080', method='GET', uri='/rest/all', version='HTTP/1.1', remote_ip='127.0.0.1', headers={'Host': '127.0.0.1:8080', 'User-Agent': 'Wget/1.18 (linux-gnueabihf)', 'Connection': 'Keep-Alive', 'Accept': '*/*', 'Accept-Encoding': 'identity'})
    Traceback (most recent call last):
      File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 1509, in _execute
        result = method(*self.path_args, **self.path_kwargs)
      File "/opt/evok/evok.py", line 1556, in get
        result += map(lambda dev: dev.full(), Devices.by_int(WATCHDOG))
      File "/opt/evok/evok.py", line 1556, in <lambda>
        result += map(lambda dev: dev.full(), Devices.by_int(WATCHDOG))
      File "/opt/evok/neuron.py", line 849, in full
        return {'dev': 'wd', 'circuit': self.circuit, 'value': self.regvalue(), 'timeout': self.timeoutvalue()}
      File "/opt/evok/neuron.py", line 842, in <lambda>
        self.timeoutvalue = lambda: arm.configs[timeout_reg]
    IndexError: list index out of range
[E 171002 09:50:29 web:2063] 500 GET /rest/all (127.0.0.1) 10.25ms
```

tknot, 2017-10-02:
Ahh! It looks like the hardware definitions haven't been properly updated in the current beta release commit. I've added them in ( 63821a1214f4e0d82f69d189cfb388e99ea40f49 ) This should solve both the current and previous issues. It should be enough to download the /etc/hw_definitions/xS40.yaml and copy it over the file on you neuron device.

The definition format will be slightly different in the next beta, so this will have no bearing on it.

Apologies for the issues with the extensions, the implementation was not properly updated from the last beta commit. Hopefully things will be fine now!

closed 2017-10-09 by tknot