UnipiTechnology/evok #41: Evok on L503 doesn't work (Internal error)

actiwerks, 2017-11-09:
Followed the installation instructions in the readme and all I get as any response to the rest API is 500 Internal error. Also it is not possible to stop the service using the commant from the documentation. See more details in this post : https://forum.unipi.technology/topic/488/internal-error-500-after-fresh-evok-installation

tknot, 2017-11-10:
As per the reply on the forums, the command doesn't work because you lack root privileges when running it, which you can fix by running "sudo su" beforehand. As for the first issue, it looks like EVOK cannot see your Neuron for some reason.

It might be easier to debug this on the forum instead, before we can get down to why it happens exactly. Our reference development device in fact is Neuron L503, which means that the error is very unlikely to apply universally to the L503 model.

tknot, 2017-11-10:
It looks like this might be down to a faulty installation; I have just now installed EVOK on fresh Raspbian Stretch / L503 "2017-09-07-raspbian-stretch.img", and it is working as it is supposed to; reinstalling should therefore be able to fix the issue.

tknot, 2017-11-28:
Considered solved until further information

closed 2017-11-28 by tknot

actiwerks, 2017-11-28:
Yeah, thanks for the tip, I reinstalled the latest image and now my device works. You were faster than me to close the ticket. Thanks for your help.