UnipiTechnology/evok #42: [NEURON] 'NoneType' object has no attribute 'do_scan'

phillipsnick, 2017-11-23:
### Prerequisites

* [ ] Can you reproduce the problem in safe mode?
Untested

* [ ] Are you running the latest Master or the latest release version?
Tried latest master and release v2.0a

* [] Did you try to reinstall EVOK?
Yes and a fresh install of Raspbian Stretch Lite

Version:September 2017
Release date:2017-09-07
Kernel version:4.9

* [ ] Did you perform a cursory search on the [forum] and [google]?

### Log files needed

Include the output of the following commands on your Neuron/UniPi device:

/var/log/evok.log

```
2017-11-23 23:19:36,917 - evok - ERROR - 'model'
Traceback (most recent call last):
  File "/opt/evok/neuron.py", line 185, in readboards
    yield board.parse_definition(self.hw_dict, i)
  File "/usr/local/lib/python2.7/dist-packages/tornado/gen.py", line 1055, in run
    value = future.result()
  File "/usr/local/lib/python2.7/dist-packages/tornado/concurrent.py", line 238, in result
    raise_exc_info(self._exc_info)
  File "/usr/local/lib/python2.7/dist-packages/tornado/gen.py", line 307, in wrapper
    yielded = next(result)
  File "/opt/evok/neuron.py", line 574, in parse_definition
    if defin and defin['type'] in config.globals['model']:
KeyError: 'model'
2017-11-23 23:19:36,924 - evok - ERROR - 'model'
Traceback (most recent call last):
  File "/opt/evok/neuron.py", line 185, in readboards
    yield board.parse_definition(self.hw_dict, i)
  File "/usr/local/lib/python2.7/dist-packages/tornado/gen.py", line 1055, in run
    value = future.result()
  File "/usr/local/lib/python2.7/dist-packages/tornado/concurrent.py", line 238, in result
    raise_exc_info(self._exc_info)
  File "/usr/local/lib/python2.7/dist-packages/tornado/gen.py", line 307, in wrapper
    yielded = next(result)
  File "/opt/evok/neuron.py", line 574, in parse_definition
    if defin and defin['type'] in config.globals['model']:
KeyError: 'model'
2017-11-23 23:19:36,928 - evok - ERROR - Invalid device circuit number 1_01
Traceback (most recent call last):
  File "/opt/evok/config.py", line 353, in add_aliases
    dev_obj = Devices.by_int(dev_pointer["dev_type"], dev_pointer["circuit"])
  File "/opt/evok/devices.py", line 67, in by_int
    raise Exception('Invalid device circuit number %s' % str(circuit))
Exception: Invalid device circuit number 1_01
2017-11-23 23:19:36,929 - evok - ERROR - Invalid device circuit number 1_02
Traceback (most recent call last):
  File "/opt/evok/config.py", line 353, in add_aliases
    dev_obj = Devices.by_int(dev_pointer["dev_type"], dev_pointer["circuit"])
  File "/opt/evok/devices.py", line 67, in by_int
    raise Exception('Invalid device circuit number %s' % str(circuit))
Exception: Invalid device circuit number 1_02
2017-11-23 23:19:37,395 - evok - ERROR - 'NoneType' object has no attribute 'do_scan'
Traceback (most recent call last):
  File "/opt/evok/neuron.py", line 225, in scan_boards
    yield self.modbus_cache_map.do_scan()
AttributeError: 'NoneType' object has no attribute 'do_scan'
2017-11-23 23:19:37,897 - evok - ERROR - 'NoneType' object has no attribute 'do_scan'
[... 1569 lines cut ...]
      File "/opt/evok/neuron.py", line 225, in scan_boards
        yield self.modbus_cache_map.do_scan()
    AttributeError: 'NoneType' object has no attribute 'do_scan'
[E 171123 23:22:16 neuron:227] 'NoneType' object has no attribute 'do_scan'
    Traceback (most recent call last):
      File "/opt/evok/neuron.py", line 225, in scan_boards
        yield self.modbus_cache_map.do_scan()
    AttributeError: 'NoneType' object has no attribute 'do_scan'
[E 171123 23:22:16 neuron:227] 'NoneType' object has no attribute 'do_scan'
    Traceback (most recent call last):
      File "/opt/evok/neuron.py", line 225, in scan_boards
        yield self.modbus_cache_map.do_scan()
    AttributeError: 'NoneType' object has no attribute 'do_scan'
^C[I 171123 23:22:16 evok:1779] Shutting down
```

### Description

Service never starts due to error on https://github.com/UniPiTechnology/evok/blob/master/evok/neuron.py#L225

### Steps to Reproduce

Install fresh Raspbian
Install Evok as per documentation at https://evok.api-docs.io/1.0/jKcTKe5aRBCNjt8Az/introduction
Run in debug mode

tknot, 2017-11-24:
It looks like your I2C EEPROM is not available (and EVOK therefore cannot tell which Neuron you have), could you post the contents of you 
`/sys/class/i2c-dev/i2c-1/device/1-0050/eeprom` 
file? (i.e. by using `cat /sys/class/i2c-dev/i2c-1/device/1-0050/eeprom`)

This error usually indicates that the installation didn't go through correctly. Unfortunately we haven't yet been able to replicate the error ourselves without turning off I2C in /boot/config.txt. It would help if we could know the output from:
`lsmod`,
`i2cdetect 1`,
and 
`cat /boot/config.txt`

tknot, 2017-11-24:
The latest commit (5271b84c56c7facc17797a47e3824cf86efc485d) should make the I2C handling more robust. It's possible that I2C-dev did not enable correctly, the latest version should fix that.

phillipsnick, 2017-11-24:
Release v2.0.1 has solved this issue, thanks!

aufi, 2018-12-25:
Hi, I experience the same error after updating to latest evok with apt upgrade (and other related packages, run apt upgrade today and reboot, see fragment bellow).

I tried replace my current /etc/evok.conf file with repo's evok-neuron.conf and no change. Installed evok to raspbian stretch with repo from apt repository.

Any idea where to debug/fix it? Thanks!

```
...
Removing 'diversion of /boot/overlays/ds2482.dtbo to /usr/share/rpikernelhack/overlays/ds2482.dtbo by rpikernelhack'
Removing 'diversion of /boot/overlays/neuronee.dtbo to /usr/share/rpikernelhack/overlays/neuronee.dtbo by rpikernelhack'
Removing 'diversion of /boot/overlays/neuron-spi-new.dtbo to /usr/share/rpikernelhack/overlays/neuron-spi-new.dtbo by rpikernelhack'
Removing 'diversion of /boot/overlays/neuron-spi.dtbo to /usr/share/rpikernelhack/overlays/neuron-spi.dtbo by rpikernelhack'
Removing 'diversion of /boot/overlays/unipiee.dtbo to /usr/share/rpikernelhack/overlays/unipiee.dtbo by rpikernelhack'
OK
update-initramfs: Generating /boot/initrd.img-4.14.71-v7+
WARNING: missing /lib/modules/4.14.71-v7+
Ensure all necessary drivers are built into the linux image!
depmod: ERROR: could not open directory /lib/modules/4.14.71-v7+: No such file or directory
depmod: FATAL: could not search modules: No such file or directory
depmod: WARNING: could not open /var/tmp/mkinitramfs_opXti4/lib/modules/4.14.71-v7+/modules.order: No such file or directory
depmod: WARNING: could not open /var/tmp/mkinitramfs_opXti4/lib/modules/4.14.71-v7+/modules.builtin: No such file or directory
Setting up libtiff5:armhf (4.0.8-2+deb9u4) ...
Setting up perl (5.24.1-3+deb9u5) ...
Setting up libssl1.0.2:armhf (1.0.2q-1~deb9u1) ...
Processing triggers for libc-bin (2.24-11+deb9u3) ...
Setting up unipi-modbus-tools (1.1.14) ...
Installing new version of config file /etc/default/unipi-modbus-tools ...
Setting up libssl1.1:armhf (1.1.0j-1~deb9u1) ...
Setting up openssl (1.1.0j-1~deb9u1) ...
Processing triggers for man-db (2.7.6.1-2) ...
Processing triggers for dbus (1.10.26-0+deb9u1) ...
Setting up evok (2.0.7d) ...
Setting up libpolkit-gobject-1-0:armhf (0.105-18+deb9u1) ...
```

```
root@neuronl203:/etc# cat /boot/config.txt
dtparam=i2c_arm=on
dtoverlay=neuronee
dtoverlay=i2c-rtc,mcp7941x
dtoverlay=unipiee
dtoverlay=neuron-spi-new
dtparam=i2c_arm=on
dtoverlay=neuronee
dtoverlay=i2c-rtc,mcp7941x
dtoverlay=unipiee
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
[... 83 lines cut ...]
w1_gpio                16384  0
wire                   40960  1 w1_gpio
cn                     16384  1 wire
hwmon                  16384  2 rtc_ds1307,wire
uio_pdrv_genirq        16384  0
fixed                  16384  0
uio                    20480  1 uio_pdrv_genirq
ip_tables              24576  0
x_tables               32768  1 ip_tables
ipv6                  425984  32
```

Current error in log:
```
2018-12-25 23:03:06,048 - evok - ERROR - 'NoneType' object has no attribute 'do_scan'
Traceback (most recent call last):
  File "/opt/evok/lib/python2.7/site-packages/evok/neuron.py", line 269, in scan_boards
    yield self.modbus_cache_map.do_scan()
AttributeError: 'NoneType' object has no attribute 'do_scan'
2018-12-25 23:03:06,150 - evok - ERROR - 'NoneType' object has no attribute 'do_scan'
Traceback (most recent call last):
  File "/opt/evok/lib/python2.7/site-packages/evok/neuron.py", line 269, in scan_boards
    yield self.modbus_cache_map.do_scan()
AttributeError: 'NoneType' object has no attribute 'do_scan'
```

tknot, 2018-12-26:
This looks like an issue with installing the unipi driver kernel module, possibly due to kernel version incompatibility. The module is not listed on lsmod and so is not loaded, and it is an essential component for any communication with Neuron or Axon.

I am away from the office during the christmas period and so cannot give a comprehensive answer, but I can see two possible solutions without further testing -
 reinstalling the kernel module by first executing 'apt-get remove unipi-kernel --purge' and then installing the dynamically compiled version with 'apt-get install unipi-kernel-dkms'. 

Unfortunately I cannot guarantee that this will work without further errors, as I do not know the complete state of the device. The other option would be using a clean image, though if you are willing to wait I may be abpe to suggest a better solution after the christmas period.

aufi, 2018-12-26:
Thanks for very quick response! I will try reinstall kernel and provide update (or use clean image if it fails, actually I use it as lights control in my house:-)). I'm happy discuss it after Christmas.

aufi, 2018-12-31:
It haven't booted after kernel reinstall, so used a new card with fresh image (but I have the original memory card for possible further investigation).

tknot, 2019-01-04:
Very odd to hear it hasn't booted. If you want you can upload an image of the sd card to mediafire.com 
and I'll have a look at it, but I can't promise I'll be able to get to the bottom of it.

Regardless I'm glad to hear it works with a fresh image. The kernel upgrade process on Linux is unfortunately prone to errors compared to other OS, it's a common problem with other 3rd-party kernel drivers, e.g. virtualbox.

aufi, 2019-03-30:
I'm sorry I don't have the card anymore, but I experienced similar issue, which was solved by evok reinstall (which rebuilt kernel modules).

I think the issue can be closed, thanks.

closed 2024-03-27 by kratochvil01