UnipiTechnology/evok #46: Is it normal that "DS2482-100 bus master" keeps reconnecting

eLvErDe, 2017-12-22:
Hello,

Evok keeps showing the following log. I'm running 2.0.1 on Neuron devices.

Is this something I should worry about ? Does it means there's something improperly setup ?

```
Dec 22 12:45:44 cara01 OWFS[597]: DEFAULT: ow_reconnect.c:(72) DS2482-100 bus master reconnected
Dec 22 12:46:04 cara01 OWFS[597]: DEFAULT: ow_reconnect.c:(72) DS2482-100 bus master reconnected
Dec 22 12:46:24 cara01 OWFS[597]: DEFAULT: ow_reconnect.c:(72) DS2482-100 bus master reconnected
Dec 22 12:46:44 cara01 OWFS[597]: DEFAULT: ow_reconnect.c:(72) DS2482-100 bus master reconnected
Dec 22 12:46:44 cara01 OWFS[597]: DEFAULT: ow_reconnect.c:(72) DS2482-100 bus master reconnected
Dec 22 12:47:04 cara01 OWFS[597]: DEFAULT: ow_reconnect.c:(72) DS2482-100 bus master reconnected
Dec 22 12:47:24 cara01 OWFS[597]: DEFAULT: ow_reconnect.c:(72) DS2482-100 bus master reconnected
```

Thanks in advance,

Adam.

tknot, 2018-01-02:
Hi Adam,
The messages should be harmless, and are due to the recently added support for CODESYS, which is done by wrapping the 1Wire bus master in another device. The messages are a side effect of this. Unfortunately it would be difficult for us to suppress them, as they originated from OWFS itself.

We might look into it further at some point, but for now it has low priority for us.

sushukka, 2020-02-06:
Should this be fixed by now? I'm running L503s and the flooding of these messages in journalctl are just not the way it should be. It's hard to pick up important messages when 80% of the messages is now "OWFS[1120]: DEFAULT: ow_reconnect.c:(72) DS2482-100 bus master reconnected". I don't specially even need CODESYS so could this be disabled somewhere or could you finally fix this minor but quite annoying issue, please?

martytriska, 2020-02-07:
Hello @sushukka,

we can revise the issue and try to fix it. As mentioned @tknot, this message comes from 3rd party software product (OWFS) and should be suppressed. So, I cannot guarantee the fix in the next version (released in a few days) as there will be probably more work than it seems on it. Thank you for understanding.

sushukka, 2020-02-07:
Hello @martytriska and thank you for the quick answer. Could OWFS be informed or asked to give some suppression option for this? 

Of course this is not a major issue and you have probably more important bugs to fix, but imho in industry level solutions like UniPi products are (not some random Raspberry gizmos), these should be eventually be ironed out. We're now in alpha-production phase and need often rely on journal/logs for debugging/finetuning and these message are starting to be quite annoying. Soon we'll have dozens of UniPis in production and luckily this seems to be now the only nuisance we would like to get fixed. So in overall good work. :)

martytriska, 2020-02-11:
Hello @sushukka,

yes, I definitely agree with you. We are going to focus on the issue in the upcoming release. Thank you for the compliment :). In the meantime, the issue here remains open.

sushukka, 2020-04-05:
Seems that you have fixed it. After updating couple of UniPi Neurons couple of days ago, it seems that these messages don't appear anymore on syslog. Superb!

martytriska, 2020-04-06:
Hello @sushukka,

you are right. We have done some changes in access to the I2C bus (another Kernel module is used now).  
So, as the side effect, the message mentioned may not be logged anymore.

closed 2020-04-06 by martytriska