UnipiTechnology/evok #48: 1Wire 4-Relay + 4-DI extension not working with Neuron Evok API

dgajic, 2018-01-17:
I've purchased 1 Wire, 4RO+4DI extension but it is not supported in the current version of Evok API. 

I've tracked the code and located issue in config.py L:209:
`r = unipig.DS2408_relay(circuit, sensor, pin, dev_id=0)`

I assume unipig module is not loaded for Neuron device, since it contains code specific for UniPi. 

I did managed to get it working by moving code (class DS2408_pio, class DS2408_input, class DS2408_rela) to config.py, the only thing I needed to change is dev_id since if it remains 0 (the default) extension DIs overlap Neuron's DI 1_01-1_04 (I have M203). It should probably go to the configuration, I've also changed default for major_group from DS2408_pio constructor, not sure for what is it used for.

I can create pull request if you want, but I'm not sure is this the right way and also not verbose with Python so better to ask first.

tknot, 2018-01-23:
config.py is common to both Neuron and the UniPi 1.1 - UniPi 1.1 parsing is done through the same create_devices function as is Neuron. In fact most of config.py is concerned with Neuron, since only the Neuron device itself is present. Could you share the changes which you needed to make?

kratochvil01, 2024-06-20:
Pigpio has been completely removed from Evok. Evok now communicates only with Modbus RTU or Modbus TCP. Support for Unipi 1.X has been resolved using [unipi-one-modbus](https://github.com/UniPiTechnology/unipi-one-modbus).

closed 2024-06-20 by kratochvil01