UnipiTechnology/evok #49: Evok v2.0.1 rpc server not found (404)

vdwemil95, 2018-02-01:
### Prerequisites

*[]Tried to reinstall evok multiple times.

- Using the latest version (2.0.1)
- Tried to reinstall evok multiple times.
- Could not find a solution on google or the forums.

### Log files needed
Please find attached

evok.log is empty
[evok.access.log](https://github.com/UniPiTechnology/evok/files/1685145/evok.access.log)
[ps_fax_output.txt](https://github.com/UniPiTechnology/evok/files/1685147/ps_fax_output.txt)

### Description

After updating to evok from version 1.0.2 to 2.0.1 on a raspberry pi 3B (raspbian) the evok server returns 404 for any rpc requests.

### Steps to Reproduce

1. Install evok version 2.0.1 for Unipi v1.x
2. reboot
3. Do a remote procedure call to the ip and website port as set up in the installation

**Expected behavior:** 
Return the result from evok server

**Actual behavior:** 
Returns 404, 'Not Found'

[forum]:https://forum.unipi.technology/category/4/official-evok-api
[google]:http://www.google.com/

vdancsi, 2018-02-07:
Same here. When trying to start evok in debug mode, it says "nameerror: tornado not defined" 
Any idea?

vdwemil95, 2018-02-07:
Did you try to install the tornado library using pip?

Sent from ProtonMail mobile

-------- Original Message --------

vdancsi, 2018-02-07:
Yes, same result. Interesting thing is that with the same clear install procedure on another unipi that was installed one day earlier works fine, but none of the two that was installed yesterday works, same problem.

vdwemil95, 2018-02-07:
I dont know what the issue could be because the install script should install all dependancies and log it if there was an error. I checked my logs and there were no errors.

Sent from ProtonMail mobile

-------- Original Message --------

tknot, 2018-02-07:
The install error should be now fixed. We do our development on Windows which has slightly different python dependencies and therefore imports (specifically the "re" module). One of the imports got accidentally left out, but due to workarounds it worked on our development system.

Unfortunately I cannot guarantee the original error of this bug is fixed yet, but the Master EVOK branch should install correctly now.

tknot, 2018-02-07:
After more testing the issue appears to be that RPC only lives on the API port (i.e. by default 8080), rather than the website port (80 by default). In the as-of-now latest master commit (149190241bfc7a3fa81fe43b6115ea81a0a18761) this has now been changed to allow both 80 and 8080 to work.

Try using either the latest master, or using the address with the ip port specified, i.e. "127.0.0.1:8080/rpc" (assuming default ports and the RPC running on localhost) instead.