UnipiTechnology/evok #51: no boards detected after apt-get upgrade

tonytee, 2018-03-21:
### Prerequisites

* [ v] Are you running the latest Master or the latest release version?
* [ v] Did you try to reinstall EVOK?
* [ v] Did you perform a cursory search on the [forum] and [google]?

### Log files needed

Include the output of the following commands on your Neuron/UniPi device:

```
sudo su
cat /var/log/evok.access.log
ps -fax
cat /var/log/evok.log
```
root@S103-sn488:~# cat /var/log/evok.access.log
172.16.129.31 - - [21/Mar/2018:08:49:50 +0000] "GET /rest/all HTTP/1.1" 200 2644 "http://172.16.129.32/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_11_6) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/65.0.3325.162 Safari/537.36"
172.16.129.31 - - [21/Mar/2018:08:50:14 +0000] "GET /ws HTTP/1.1" 101 6570 "-" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_11_6) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/65.0.3325.162 Safari/537.36"
172.16.129.31 - - [21/Mar/2018:08:50:14 +0000] "GET / HTTP/1.1" 304 0 "-" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_11_6) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/65.0.3325.162 Safari/537.36"
172.16.129.31 - - [21/Mar/2018:08:50:15 +0000] "GET /rest/all HTTP/1.1" 304 0 "http://172.16.129.32/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_11_6) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/65.0.3325.162 Safari/537.36"

root@S103-sn488:~# ps -fax
  PID TTY      STAT   TIME COMMAND
    2 ?        S      0:00 [kthreadd]
    3 ?        S      0:00  \_ [ksoftirqd/0]
    5 ?        S<     0:00  \_ [kworker/0:0H]
    6 ?        S      0:00  \_ [kworker/u8:0]
    7 ?        S      0:00  \_ [rcu_sched]
    8 ?        S      0:00  \_ [rcu_bh]
    9 ?        S      0:00  \_ [migration/0]
   10 ?        S<     0:00  \_ [lru-add-drain]
   11 ?        S      0:00  \_ [cpuhp/0]
   12 ?        S      0:00  \_ [cpuhp/1]
   13 ?        S      0:00  \_ [migration/1]
   14 ?        S      0:00  \_ [ksoftirqd/1]
   16 ?        S<     0:00  \_ [kworker/1:0H]
   17 ?        S      0:00  \_ [cpuhp/2]
   18 ?        S      0:00  \_ [migration/2]
   19 ?        S      0:00  \_ [ksoftirqd/2]
   21 ?        S<     0:00  \_ [kworker/2:0H]
   22 ?        S      0:00  \_ [cpuhp/3]
   23 ?        S      0:00  \_ [migration/3]
   24 ?        S      0:00  \_ [ksoftirqd/3]
   26 ?        S<     0:00  \_ [kworker/3:0H]
   27 ?        S      0:00  \_ [kdevtmpfs]
   28 ?        S<     0:00  \_ [netns]
   29 ?        S      0:00  \_ [khungtaskd]
   30 ?        S      0:00  \_ [oom_reaper]
   31 ?        S<     0:00  \_ [writeback]
   32 ?        S      0:00  \_ [kcompactd0]
   33 ?        S<     0:00  \_ [crypto]
   34 ?        S<     0:00  \_ [bioset]
   35 ?        S<     0:00  \_ [kblockd]
   36 ?        S<     0:00  \_ [watchdogd]
   37 ?        S      0:00  \_ [kworker/0:1]
   38 ?        S<     0:00  \_ [rpciod]
   39 ?        S<     0:00  \_ [xprtiod]
   40 ?        S      0:00  \_ [kswapd0]
   41 ?        S<     0:00  \_ [vmstat]
   42 ?        S<     0:00  \_ [nfsiod]
   52 ?        S<     0:00  \_ [kthrotld]
   53 ?        S<     0:00  \_ [bioset]
   54 ?        S<     0:00  \_ [bioset]
   55 ?        S<     0:00  \_ [bioset]
   56 ?        S<     0:00  \_ [bioset]
   57 ?        S<     0:00  \_ [bioset]
   58 ?        S<     0:00  \_ [bioset]
   59 ?        S<     0:00  \_ [bioset]
   60 ?        S<     0:00  \_ [bioset]
[... 121 lines cut ...]
### Steps to Reproduce

1. start out with image UniPian-Neuron-OS-2018-03-09.img all works fine out of the box
Evok fine everything fine

root@S103-sn488:/boot# uname -a
Linux S103-sn488 4.9.41-v7+ #1023 SMP Tue Aug 8 16:00:15 BST 2017 armv7l GNU/Linux

2. apt-get update / apt-get upgrade

root@S103-sn488:/home/pi# uname -a
Linux S103-sn488 4.9.80-v7+ #1098 SMP Fri Mar 9 19:11:42 GMT 2018 armv7l GNU/Linux

result: No boards found on SPI 1, 2, 3

3. tried manual install, evok-2.0.3a didn't fix things

**Expected behavior:** expected to be able to run safe apt-get upgrade

**Actual behavior:** things got broken

[forum]:https://forum.unipi.technology/category/4/official-evok-api
^^posted here too, sorry if doubles are inconvenient

[google]:http://www.google.com/

tknot, 2018-03-22:
As with my reply on the forum, it seems likely that the kernel driver module (either neuronspi.ko or unipi.ko, depending on which distribution you have) is unable to load due to the RPi kernel update performed by apt-get upgrade. I'll test this myself further, but you _may_ be able to fix this by  moving the aforementioned file (either unipi.ko or neuronspi.ko, depending on the image version you have) from 
`
/lib/modules/[the oldest version in the folder]-v7+/kernel/drivers/spi/
`
to
`
/lib/modules/[the newest version in the folder]-v7+/kernel/drivers/spi/
`

This isn't guaranteed to work as I haven't tested it with the latest update to the kernel, but it is likely to.

More generally, this issue arises from the fact that linux kernel drivers are always compiled against a specific kernel version, unlike e.g. on windows. Most universal drivers come included with the kernel, while large corporations such as e.g. NVidia create a binary version for each kernel version which gets included in common distributions. This is unfortunately not feasible for us.

Within the Linux community the "_intended_" approach is to compile the kernel module yourself; this is however not entirely trivial. You can find the source code for our kernel modules at [http://git.unipi.technology:3000/UniPi/neuron-kernel](http://git.unipi.technology:3000/UniPi/neuron-kernel)

tknot, 2018-03-22:
We should be able to fix this by creating a dummy .deb package dependent on the kernel, which will prevent it from being updated by apt-get upgrade.

tonytee, 2018-03-22:
Thanks for your reply, 

I have tried your suggestion by copying
/lib/modules/4.9.41-v7+/kernel/drivers/spi/neuronspi.ko
to both
/lib/modules/4.9.80-v7+/kernel/drivers/spi/
and
/lib/modules/4.9.80+/kernel/drivers/spi/

but sofar no fix... it could be that i broke too much on the image by now, so i need to try this again on a clean one, with a clean apt-get update/upgrade.

also, maybe some other .ko's are not playing nicely
in the newer spi folders next to the "old"
-rw-r--r-- 1 root root 41226 Mar 22 17:58 neuronspi.ko
 i also have
-rw-r--r-- 1 root root 14416 Mar 13 20:53 spi-bcm2835.ko
-rw-r--r-- 1 root root 11152 Mar 13 20:53 spi-bcm2835aux.ko
-rw-r--r-- 1 root root 16072 Mar 13 20:53 spidev.ko

if this quick fix neither works on the clean image i'll give the module compile a go.

i'll get back to you with my findings on both routes tomorrow...

thanks, tony

yaneti, 2018-03-22:
> Within the Linux community the 'intended' approach is to compile the kernel module yourself; 

Nope , the intention is for all hardware manufacturers to see the light and get their kernel code upstream. ;)

tknot, 2018-03-22:
Nope , the intention is for all hardware manufacturers to see the light and get their kernel code upstream. ;)

Point taken :) Our drivers don't really fit within the established framework of kernel drivers unfortunately though, some of it is due to business decisions (the multiple levels of platform devices for one) and some of it is due to the nature of the device. One could note that no other PLC manufacturer has drivers within the kernel, despite us knowing of several which use mainstream Linux.

And, well, they would need to be cleaned up quite a bit before they could get upstream.

tonytee, 2018-03-23:
Hi Tomas,

Sorry to say that I can now confirm that the "moving neuronspi.ko to new location fix" does not work on a clean install either.

that is, starting out with UniPian-Neuron-OS-2018-03-09.img, seeying taht it actually works, and a apt-get update/upgrade right after that.

I haven't put much time in the module recompile, couldn't get it to work... as you said, not so trivial. 
At least not to me :)

My conclusion is that there's not an easy fix for rescuing a broken system, so just sheer providence made that i luckily didn't need that fix anywhere (yet)

However I also looked into your package pinning suggestion.

apt-mark hold raspberrypi-bootloader
apt-mark hold python-tornado
apt-mark hold raspberrypi-kernel

before an upgrade at least prevents things getting broken

so at least there's a temporary prevention scheme

cheers, please let me know if i can help more

tony

tknot, 2018-04-03:
It looks like we'll have to pin the packages on our images as well, at least for now. Our .deb package repository is taking longer to setup than we anticipated, as we have decided to include CI and all of our other software as well.

We will try to keep the packages pinned by default in the next image version, and fix this properly once we have our repository up and running properly.

tknot, 2018-04-30:
The package system rollout should happen within the next week or two, barring any unforeseen circumstances. Internal testing is going well, the package repository will be on files.unipi.technology.

tknot, 2018-09-21:
As the package system is now active it should be possible to perform apt-get upgrade freely, provided that EVOK is installed via a debian package.

closed 2018-09-21 by tknot

daankets, 2019-04-12:
@tknot Today, I updated my UniPI Neuron to the latest raspberry pi kernel using `sudo apt-get upgrade`... The result:
```
The following packages have unmet dependencies:
 neuron-kernel : Depends: raspberrypi-kernel (= 1.20181112-1) but 1.20190401-1 is to be installed
E: Unable to correct problems, you have held broken packages.```

And therefore, Evok lists 0 devices (as the neuron kernel is missing). So it seems this issue is not yet resolved?

tomashora, 2019-04-30:
That is a general dependency issue. After RPi releases a new kernel version, we have to test everything and release a new version of our kernel drivers which usually takes some time. If users install/update the RPi kernel before we release the new compatible kernel driver, the install fails which causes malfunctional Evok.
@daankets That is fixed after the release of neuron-kernel: 1.22.1.20190401-1 and unipi-kernel-modules-dkms: 1.22 Please apt update & upgrade.