UnipiTechnology/evok #52: Wrong count of I/O on Neuron M203

eLvErDe, 2018-03-21:
Hello,

I'm having an issue with M203, one of them is not reporting proper count of I/O in Evok REST API.

This one is working properly
```
Mar 21 10:54:26 w1e1w1c11100245 evok[8486]: [I 180321 10:54:26 config:66] eprom: UniPi Neuron M203 version: 1.0 serial: 0x4b
Mar 21 10:54:26 w1e1w1c11100245 evok[8486]: [I 180321 10:54:26 evok:1640] Starting using config file /etc/evok.conf
Mar 21 10:54:27 w1e1w1c11100245 evok[8486]: [I 180321 10:54:27 config:78] YAML Definition loaded: DALI.yaml, type: 3, definition count 0
Mar 21 10:54:27 w1e1w1c11100245 evok[8486]: [I 180321 10:54:27 config:78] YAML Definition loaded: S10x.yaml, type: 3, definition count 1
Mar 21 10:54:27 w1e1w1c11100245 evok[8486]: [I 180321 10:54:27 config:78] YAML Definition loaded: xS40.yaml, type: 3, definition count 2
Mar 21 10:54:27 w1e1w1c11100245 evok[8486]: [I 180321 10:54:27 config:78] YAML Definition loaded: xS50.yaml, type: 3, definition count 3
Mar 21 10:54:27 w1e1w1c11100245 evok[8486]: [I 180321 10:54:27 config:78] YAML Definition loaded: L20x.yaml, type: 3, definition count 4
Mar 21 10:54:27 w1e1w1c11100245 evok[8486]: [I 180321 10:54:27 config:78] YAML Definition loaded: L50x.yaml, type: 3, definition count 5
Mar 21 10:54:28 w1e1w1c11100245 evok[8486]: [I 180321 10:54:28 config:78] YAML Definition loaded: L40x.yaml, type: 3, definition count 6
Mar 21 10:54:28 w1e1w1c11100245 evok[8486]: [I 180321 10:54:28 config:78] YAML Definition loaded: xS30.yaml, type: 3, definition count 7
Mar 21 10:54:28 w1e1w1c11100245 evok[8486]: [I 180321 10:54:28 config:78] YAML Definition loaded: L30x.yaml, type: 3, definition count 8
Mar 21 10:54:28 w1e1w1c11100245 evok[8486]: [I 180321 10:54:28 config:78] YAML Definition loaded: M30x.yaml, type: 3, definition count 9
Mar 21 10:54:28 w1e1w1c11100245 evok[8486]: [I 180321 10:54:28 config:78] YAML Definition loaded: xS10.yaml, type: 3, definition count 10
Mar 21 10:54:29 w1e1w1c11100245 evok[8486]: [I 180321 10:54:29 config:78] YAML Definition loaded: L51x.yaml, type: 3, definition count 11
Mar 21 10:54:29 w1e1w1c11100245 evok[8486]: [I 180321 10:54:29 config:78] YAML Definition loaded: M50x.yaml, type: 3, definition count 12
Mar 21 10:54:29 w1e1w1c11100245 evok[8486]: [I 180321 10:54:29 config:78] YAML Definition loaded: M40x.yaml, type: 3, definition count 13
Mar 21 10:54:29 w1e1w1c11100245 evok[8486]: [I 180321 10:54:29 config:78] YAML Definition loaded: CUSTOM MODBUS DEVICE.yaml, type: 3, definition count 14
Mar 21 10:54:29 w1e1w1c11100245 evok[8486]: [I 180321 10:54:29 config:78] YAML Definition loaded: M10x.yaml, type: 3, definition count 15
Mar 21 10:54:29 w1e1w1c11100245 evok[8486]: [I 180321 10:54:29 config:78] YAML Definition loaded: M20x.yaml, type: 3, definition count 16
Mar 21 10:54:29 w1e1w1c11100245 evok[8486]: [I 180321 10:54:29 config:78] YAML Definition loaded: evok-alias.yaml, type: 2, definition count 0
Mar 21 10:54:29 w1e1w1c11100245 evok[8486]: [I 180321 10:54:29 evok:1704] HTTP server listening on port: 8080
Mar 21 10:54:30 w1e1w1c11100245 evok[8486]: [I 180321 10:54:30 modbusclient_tornado:125] SPI client started
Mar 21 10:54:30 w1e1w1c11100245 evok[8486]: [I 180321 10:54:30 neuron:173] Reading SPI boards
Mar 21 10:54:30 w1e1w1c11100245 evok[8486]: [D 180321 10:54:30 owclient:376] Entering 1wire loop
Mar 21 10:54:30 w1e1w1c11100245 evok[8486]: [I 180321 10:54:30 neuron:188] No board on SPI 3
Mar 21 10:54:30 w1e1w1c11100245 evok[8486]: [I 180321 10:54:30 config:354] Alias loaded: <neuron.Relay object at 0x7604dc10> al_lights_kitchen
Mar 21 10:54:30 w1e1w1c11100245 evok[8486]: [I 180321 10:54:30 config:354] Alias loaded: <neuron.Relay object at 0x7604d6b0> al_lights_bedroom
```
The other one reporting limited count of I/O
```
Nov  3 17:16:47 neuron evok[336]: [I 161103 17:16:47 config:66] eprom: UniPi Neuron M203 version: 1.0 serial: 0x4a
Nov  3 17:16:47 neuron evok[336]: [I 161103 17:16:47 evok:1640] Starting using config file /etc/evok.conf
Nov  3 17:16:47 neuron evok[336]: [I 161103 17:16:47 config:78] YAML Definition loaded: L20x.yaml, type: 3, definition count 0
Nov  3 17:16:47 neuron evok[336]: [I 161103 17:16:47 config:78] YAML Definition loaded: L30x.yaml, type: 3, definition count 1
Nov  3 17:16:47 neuron evok[336]: [I 161103 17:16:47 config:78] YAML Definition loaded: S10x.yaml, type: 3, definition count 2
Nov  3 17:16:47 neuron evok[336]: [I 161103 17:16:47 config:78] YAML Definition loaded: L40x.yaml, type: 3, definition count 3
Nov  3 17:16:48 neuron evok[336]: [I 161103 17:16:48 config:78] YAML Definition loaded: M50x.yaml, type: 3, definition count 4
Nov  3 17:16:48 neuron evok[336]: [I 161103 17:16:48 config:78] YAML Definition loaded: xS10.yaml, type: 3, definition count 5
Nov  3 17:16:48 neuron evok[336]: [I 161103 17:16:48 config:78] YAML Definition loaded: xS50.yaml, type: 3, definition count 6
Nov  3 17:16:48 neuron evok[336]: [I 161103 17:16:48 config:78] YAML Definition loaded: L50x.yaml, type: 3, definition count 7
Nov  3 17:16:49 neuron evok[336]: [I 161103 17:16:49 config:78] YAML Definition loaded: L51x.yaml, type: 3, definition count 8
Nov  3 17:16:49 neuron evok[336]: [I 161103 17:16:49 config:78] YAML Definition loaded: CUSTOM MODBUS DEVICE.yaml, type: 3, definition count 9
Nov  3 17:16:49 neuron evok[336]: [I 161103 17:16:49 config:78] YAML Definition loaded: M40x.yaml, type: 3, definition count 10
Nov  3 17:16:49 neuron evok[336]: [I 161103 17:16:49 config:78] YAML Definition loaded: M20x.yaml, type: 3, definition count 11
Nov  3 17:16:49 neuron evok[336]: [I 161103 17:16:49 config:78] YAML Definition loaded: M10x.yaml, type: 3, definition count 12
Nov  3 17:16:49 neuron evok[336]: [I 161103 17:16:49 config:78] YAML Definition loaded: DALI.yaml, type: 3, definition count 13
Nov  3 17:16:49 neuron evok[336]: [I 161103 17:16:49 config:78] YAML Definition loaded: xS40.yaml, type: 3, definition count 14
Nov  3 17:16:50 neuron evok[336]: [I 161103 17:16:50 config:78] YAML Definition loaded: xS30.yaml, type: 3, definition count 15
Nov  3 17:16:50 neuron evok[336]: [I 161103 17:16:50 config:78] YAML Definition loaded: M30x.yaml, type: 3, definition count 16
Nov  3 17:16:50 neuron evok[336]: [I 161103 17:16:50 config:78] YAML Definition loaded: evok-alias.yaml, type: 2, definition count 0
Nov  3 17:16:50 neuron evok[336]: [I 161103 17:16:50 evok:1704] HTTP server listening on port: 8080
Nov  3 17:16:50 neuron evok[336]: [I 161103 17:16:50 modbusclient_tornado:125] SPI client started
Nov  3 17:16:50 neuron evok[336]: [I 161103 17:16:50 neuron:173] Reading SPI boards
Nov  3 17:16:50 neuron evok[336]: [D 161103 17:16:50 owclient:376] Entering 1wire loop
Nov  3 17:16:50 neuron evok[336]: [I 161103 17:16:50 neuron:188] No board on SPI 3
Nov  3 17:16:50 neuron evok[336]: [I 161103 17:16:50 config:354] Alias loaded: <neuron.Relay object at 0x750472b0> al_lights_kitchen
Nov  3 17:16:50 neuron evok[336]: [I 161103 17:16:50 config:354] Alias loaded: <neuron.Relay object at 0x75047770> al_lights_bedroom
```

Difference spotted:

```
Nov  3 17:16:47 neuron evok[336]: [I 161103 17:16:47 config:66] eprom: UniPi Neuron M203 version: 1.0 serial: 0x4a
Nov  3 17:16:49 neuron evok[336]: [I 161103 17:16:49 config:78] YAML Definition loaded: CUSTOM MODBUS DEVICE.yaml, type: 3, definition count 9
```

vs

```
Mar 21 10:54:26 w1e1w1c11100245 evok[8486]: [I 180321 10:54:26 config:66] eprom: UniPi Neuron M203 version: 1.0 serial: 0x4b
Mar 21 10:54:29 w1e1w1c11100245 evok[8486]: [I 180321 10:54:29 config:78] YAML Definition loaded: CUSTOM MODBUS DEVICE.yaml, type: 3, definition count 14
```

Any help would be appreciated,

Best regards, Adam.

tknot, 2018-03-22:
What specifically is the wrong count? If it is 14->16 or 16->14 then it could be an error in the hardware definition, though from a brief examination I wasn't able to find one.

 If it is limited to the first group (i.e.  4/4) then that would indicate a different kind of a problem.

tknot, 2018-04-11:
After recent revisions of hardware definitions I haven't been able to reproduce this problem. Please reopen this issue if it persists.

closed 2018-04-11 by tknot