UnipiTechnology/evok #54: [NEURON] Modbus broken

c0fec0de, 2018-04-04:
### Prerequisites

* [x] Are you running the latest Master or the latest release version?
* [x] Did you try to reinstall EVOK?
* [x] Did you perform a cursory search on the [forum] and [google]?

### Log files needed

Everything is running as expected

```
 PID TTY      STAT   TIME COMMAND
    2 ?        S      0:00 [kthreadd]
    3 ?        S      0:00  \_ [ksoftirqd/0]
    5 ?        S<     0:00  \_ [kworker/0:0H]
    7 ?        S      0:00  \_ [rcu_sched]
    8 ?        S      0:00  \_ [rcu_bh]
    9 ?        S      0:00  \_ [migration/0]
   10 ?        S<     0:00  \_ [lru-add-drain]
   11 ?        S      0:00  \_ [cpuhp/0]
   12 ?        S      0:00  \_ [cpuhp/1]
   13 ?        S      0:00  \_ [migration/1]
   14 ?        S      0:00  \_ [ksoftirqd/1]
   16 ?        S<     0:00  \_ [kworker/1:0H]
   17 ?        S      0:00  \_ [cpuhp/2]
   18 ?        S      0:00  \_ [migration/2]
   19 ?        S      0:00  \_ [ksoftirqd/2]
   21 ?        S<     0:00  \_ [kworker/2:0H]
   22 ?        S      0:00  \_ [cpuhp/3]
   23 ?        S      0:00  \_ [migration/3]
   24 ?        S      0:00  \_ [ksoftirqd/3]
   26 ?        S<     0:00  \_ [kworker/3:0H]
   27 ?        S      0:00  \_ [kdevtmpfs]
   28 ?        S<     0:00  \_ [netns]
   29 ?        S      0:00  \_ [khungtaskd]
   30 ?        S      0:00  \_ [oom_reaper]
   31 ?        S<     0:00  \_ [writeback]
   32 ?        S      0:00  \_ [kcompactd0]
   33 ?        S<     0:00  \_ [crypto]
   34 ?        S<     0:00  \_ [bioset]
   35 ?        S<     0:00  \_ [kblockd]
   36 ?        S<     0:00  \_ [watchdogd]
   37 ?        S      0:00  \_ [kworker/0:1]
   38 ?        S<     0:00  \_ [rpciod]
   39 ?        S<     0:00  \_ [xprtiod]
   40 ?        S      0:00  \_ [kswapd0]
   41 ?        S<     0:00  \_ [vmstat]
   42 ?        S<     0:00  \_ [nfsiod]
   52 ?        S<     0:00  \_ [kthrotld]
   53 ?        S<     0:00  \_ [bioset]
   54 ?        S<     0:00  \_ [bioset]
   55 ?        S      0:00  \_ [kworker/1:1]
   56 ?        S<     0:00  \_ [bioset]
   57 ?        S<     0:00  \_ [bioset]
   58 ?        S<     0:00  \_ [bioset]
   59 ?        S<     0:00  \_ [bioset]
   60 ?        S<     0:00  \_ [bioset]
   61 ?        S<     0:00  \_ [bioset]
   62 ?        S<     0:00  \_ [bioset]
   63 ?        S<     0:00  \_ [bioset]
   64 ?        S<     0:00  \_ [bioset]
   65 ?        S<     0:00  \_ [bioset]
   66 ?        S<     0:00  \_ [bioset]
   67 ?        S<     0:00  \_ [bioset]
   68 ?        S<     0:00  \_ [bioset]
   69 ?        S<     0:00  \_ [bioset]
   70 ?        S<     0:00  \_ [bioset]
   71 ?        S<     0:00  \_ [bioset]
   72 ?        S<     0:00  \_ [bioset]
   73 ?        S<     0:00  \_ [bioset]
   74 ?        S<     0:00  \_ [bioset]
   75 ?        S<     0:00  \_ [bioset]
   76 ?        S<     0:00  \_ [bioset]
   77 ?        S<     0:00  \_ [bioset]
   78 ?        S<     0:00  \_ [iscsi_eh]
   79 ?        S<     0:00  \_ [dwc_otg]
   80 ?        S<     0:00  \_ [DWC Notificatio]
   81 ?        S<     0:00  \_ [VCHIQ-0]
   82 ?        S<     0:00  \_ [VCHIQr-0]
   83 ?        S<     0:00  \_ [VCHIQs-0]
   84 ?        S      0:00  \_ [VCHIQka-0]
   85 ?        S<     0:00  \_ [SMIO]
   86 ?        S      0:00  \_ [kworker/1:2]
   87 ?        S      0:00  \_ [irq/92-mmc1]
   90 ?        S<     0:00  \_ [bioset]
   91 ?        S      0:00  \_ [mmcqd/0]
   92 ?        S      0:00  \_ [kworker/2:1]
   93 ?        S      0:00  \_ [kworker/3:1]
   94 ?        S      0:00  \_ [jbd2/mmcblk0p2-]
   95 ?        S<     0:00  \_ [ext4-rsv-conver]
   96 ?        S<     0:00  \_ [ipv6_addrconf]
  108 ?        S<     0:00  \_ [kworker/2:1H]
  109 ?        S      0:00  \_ [kworker/u8:1]
  149 ?        S      0:00  \_ [kworker/3:2]
  150 ?        S      0:00  \_ [kworker/0:2]
  165 ?        S      0:00  \_ [ext4lazyinit]
  222 ?        S      0:01  \_ [spi0]
  254 ?        S<     0:00  \_ [cfg80211]
  259 ?        S<     0:00  \_ [brcmf_wq/mmc1:0]
  260 ?        S      0:00  \_ [brcmf_wdog/mmc1]
  269 ?        S      0:00  \_ [kworker/u8:2]
  271 ?        S      0:00  \_ [kworker/2:3]
  280 ?        S      0:00  \_ [neuronspi]
  288 ?        S      0:00  \_ [neuronspi]
  293 ?        S      0:00  \_ [neuronspi]
  298 ?        S      0:12  \_ [neuronspi]
  546 ?        S<     0:00  \_ [kworker/3:1H]
  556 ?        S<     0:00  \_ [kworker/1:1H]
  755 ?        S<     0:00  \_ [kworker/0:1H]
  797 ?        S      0:00  \_ [kworker/1:0]
  802 ?        S      0:00  \_ [kworker/u8:0]
  807 ?        S      0:00  \_ [kworker/0:0]
    1 ?        Ss     0:03 /sbin/init
  113 ?        Ss     0:00 /lib/systemd/systemd-journald
  152 ?        Ss     0:00 /lib/systemd/systemd-udevd
  362 ?        Ss     0:00 /opt/neuron-bin/neuron_tcp_server -p 502 -a 255 --che
  369 ?        Ss     0:00 /usr/sbin/thd --triggers /etc/triggerhappy/triggers.d
  370 ?        Ss     0:00 avahi-daemon: running [unipi-stretch.local]
  381 ?        S      0:00  \_ avahi-daemon: chroot helper
  375 ?        Ss     0:00 /usr/sbin/cron -f
  377 ?        Ss     0:00 /usr/bin/dbus-daemon --system --address=systemd: --no
  397 ?        Ssl    0:00 /usr/sbin/rsyslogd -n
  398 ?        Ss     0:00 /lib/systemd/systemd-logind
  399 ?        Ssl    0:32 /usr/bin/python /opt/evok/evok.py
  589 ?        S      0:00  \_ /usr/bin/python /opt/evok/evok.py
  458 ?        Ss     0:00 /usr/sbin/sshd -D
  760 ?        Ss     0:00  \_ sshd: pi [priv]
  777 ?        S      0:00      \_ sshd: pi@pts/0
  780 pts/0    Ss     0:00          \_ -bash
  820 pts/0    S      0:00              \_ sudo -i
  824 pts/0    S      0:00                  \_ -bash
  849 pts/0    R+     0:00                      \_ ps -fax
  471 tty1     Ss+    0:00 /sbin/agetty --noclear tty1 linux
  474 ?        Ss     0:00 /sbin/netplugd -p /var/run/netplugd.pid
  479 ?        Ss     0:00 nginx: master process /usr/sbin/nginx -g daemon on; m
  480 ?        S      0:00  \_ nginx: worker process
  481 ?        S      0:00  \_ nginx: worker process
  482 ?        S      0:00  \_ nginx: worker process
  483 ?        S      0:00  \_ nginx: worker process
  568 ?        Ssl    0:00 /lib/systemd/systemd-timesyncd
  571 ?        Ss     0:00 /sbin/dhclient -4 -v -pf /run/dhclient.eth0.pid -lf /
  767 ?        Ss     0:00 /lib/systemd/systemd --user
  770 ?        S      0:00  \_ (sd-pam)
```

`/log/evok.log` is empty

### Description

Since an update of evok, the Modbus communication is broken.

### Steps to Reproduce

Setup:
* Neuron L40x with fresh UniPian-Neuron-OS-2018-03-09.img installed
* Neuron xS30 connected via Modbus, Dip-Switch 'End' set, Address 1, Parity N

Setup works fine using Mervis image and Mervis Software with 19200/N/8.

Using UniPian OR Raspian-Lite (various versions tried), RX/TX LEDs are flashing, but off most of the time.

**Expected behavior:** Input values are properly displayed in the web interface at port 890

**Actual behavior:** No updates and errornous RX/TX lights

[forum]:https://forum.unipi.technology/category/4/official-evok-api
[google]:http://www.google.com/

tknot, 2018-04-05:
Could you try to run "sudo systemctl stop evok" and "sudo python /opt/evok.py"? The error messages may tell us more.

It's not immediately obvious to me where things are going wrong unfortunately.

c0fec0de, 2018-04-05:
Hi,

i will try out.

Are you interested in getting direct SSH access to the device?
Please send me your public SSH key and i will return the hostname and port

You should have my mail address in your history.

BR

tknot, 2018-04-05:
I will send you the key by email, but for now I have tried to replicate your setup, but I hadn't come across anything out of ordinary. Is it possible that your dip-switches are set wrong, or perhaps that your extension has not bee power-cycled?

I have used the following dip-switch settings:
[image]

,and the "UniPian-Neuron-OS-2018-03-09" image. 

Further I have used the following extension settings in evok.conf:
```
[EXTENSION_1]
global_id = 2                                                   ; Mandatory, REQUIRED TO BE UNIQUE
device_name = xS30                                              ; Mandatory
modbus_uart_port = /dev/extcomm/0/0     ; Mandatory
neuron_uart_circuit = 1_01                              ; Optional, allows associating extensions with specific Neuron UART-over-Modbus ports (not possible for non-Modbus UART ports, e.g. /dev/ttyUSB0 or /dev/ttyS0)
;allow_register_access = True                   ; Optional, False default, is mandatory with third-party devices
address = 1                                                     ; Optional, 15 default
;scan_frequency = 2                                     ; Optional, 1 default
;scan_enabled = True                                    ; Optional, True default
; Note that the following settings will be inherited by other devices sharing the same port, i.e. /dev/extcomm/0/0
;baud_rate = 19200                                              ; Optional, NEEDS UNIPI IMAGE TO WORK! USE API TO CONFIGURE UART MANUALLY IF USING STANDARD RASPBIAN
;parity = N                                                             ; Optional, NEEDS UNIPI IMAGE TO WORK! USE API TO CONFIGURE UART MANUALLY IF USING STANDARD RASPBIAN
;stop_bits = 1                                                  ; Optional, NEEDS UNIPI IMAGE TO WORK! USE API TO CONFIGURE UART MANUALLY IF USING STANDARD RASPBIAN

```
After this I have power-cycled the extension and restarted EVOK via "sudo systemctl restart evok"

tknot, 2018-04-06:
> Hi,
>
> the DIP switches are set according to the picture.
>
> When I exchange the SD with my simple mervis setup (including
> power-cycling everything), everything works.
>
> Here is the direct stdout output on the STDOUT calling evok.py directly:
>
> [I 180406 08:17:31 config:63] eprom: UniPi Neuron L403 version: 1.0 serial: 0x8
> [I 180406 08:17:31 evok:1628] Starting using config file /etc/evok.conf
> [I 180406 08:17:32 config:75] YAML Definition loaded: L20x.yaml, type: 3, definition count 0
> [I 180406 08:17:32 config:75] YAML Definition loaded: M30x.yaml, type: 3, definition count 1
> [I 180406 08:17:32 config:75] YAML Definition loaded: L50x.yaml, type: 3, definition count 2
> [I 180406 08:17:32 config:75] YAML Definition loaded: S10x.yaml, type: 3, definition count 3
> [I 180406 08:17:32 config:75] YAML Definition loaded: xS40.yaml, type: 3, definition count 4
> [I 180406 08:17:32 config:75] YAML Definition loaded: M10x.yaml, type: 3, definition count 5
> [I 180406 08:17:33 config:75] YAML Definition loaded: L30x.yaml, type: 3, definition count 6
> [I 180406 08:17:33 config:75] YAML Definition loaded: M50x.yaml, type: 3, definition count 7
> [I 180406 08:17:33 config:75] YAML Definition loaded: L51x.yaml, type: 3, definition count 8
> [I 180406 08:17:33 config:75] YAML Definition loaded: xS30.yaml, type: 3, definition count 9
> [I 180406 08:17:33 config:75] YAML Definition loaded: M40x.yaml, type: 3, definition count 10
> [I 180406 08:17:34 config:75] YAML Definition loaded: M20x.yaml, type: 3, definition count 11
> [I 180406 08:17:34 config:75] YAML Definition loaded: CUSTOM MODBUS DEVICE.yaml, type: 3, definition count 12
> [I 180406 08:17:34 config:75] YAML Definition loaded: L40x.yaml, type: 3, definition count 13
> [... 9 lines cut ...]
> [I 180406 08:17:42 web:2063] 101 GET /ws (::1) 3.90ms
> [D 180406 08:17:42 evok:158] New WebSocket client connected
> [I 180406 08:17:42 web:2063] 304 GET /rest/all (127.0.0.1) 10.78ms
>
> I also used exactly the same evok.conf lines from above

Everything looks fine in that log - which means that there are no errors,
exceptions and that the values are being read properly, because if they
were not it would cause exceptions. It also shows that no connection
attempts were made.

Could this be a browser issue? Try to use Chrome if you are not already
using it.

tknot, 2018-04-06:
Addendum to the above - it's possible that an error may happen later, as
there is a timeout of few seconds. If so then something is going wrong
communicating with the extension.

I'll send you my key for remote support if you still need to.

c0fec0de, 2018-04-06:
I think the problem is later in the communication.

I played around with baudrate and parity. They caused an exception in the logs. I also added additional debug logging after the version code reading. It seems to work.

Can we log the entire modbus communication? Can you send me a patch, which adds additional logging where modbus bytes are dumped and send me your working log for reference.

It is not a browser issue. I am using firefox and the other input ports (on the L40x itself) are signalled properly.

tknot, 2018-04-06:
Can you look what the HTTP /rest/all command is returning? It should list the UART devices circa 2-5 seconds after the main SPI device.

I can provide a patch for the modbus polling cache as well, but I would rather do this ad-hoc in a remote support session.

tknot, 2018-04-10:
As per our email conversation, this was caused by the scan_frequency configuration setting being too low to detect the short impulses on input (as anything other than counter changes, at any rate).

closed 2018-04-10 by tknot