UnipiTechnology/evok #55: webserver port

rudyvan, 2018-04-21:
I have been running a reinstall test with the latest evok, RP3 B+ and raspbian image (2018/04/18) and everything worked fine with the exception of the web server port which defaulted to 80 instead of the value 8180 that i had entered during the installation.
I installed with Unipi 1.1 

I corrected nginx/sites-enabled as per the picture below and it worked.
So for me the error is gone and i hope it is fixed when i need to install again.
 
[image]

I did not run the install again or looked for the issue in the installation script, but just want to mentioned this happened to my install.
Thanks

tknot, 2018-04-23:
It looks like this got broken in a recent commit, I'll fix it shortly

tknot, 2018-05-10:
This should be fixed since https://github.com/UniPiTechnology/evok/commit/d762f09d3485ca30b1f2fa5ff57f711bb87f753a

I forgot to update the issue earlier, apologies

tknot, 2018-05-18:
Considered solved unless further problems arise

closed 2018-05-18 by tknot