UnipiTechnology/evok #58: Internal Error response when request on non-existent device issued

jaryn, 2018-06-11:
### Log files needed
Noting in the server logs.

### Description
When issuing get request (`relay_get`) to not yet existing relay name (alias), the error -32603 is raised. I believe some other error then Internal Error should be rised. Some which would indicate that no such name of resource exists.

### Steps to Reproduce
In [1]: from jsonrpclib import Server
   ...: Server("http://192.168.1.240/rpc").relay_get('al_nonexisting')
   ...: 
---------------------------------------------------------------------------
ProtocolError                             Traceback (most recent call last)
<ipython-input-1-f614391609a1> in <module>()
      1 from jsonrpclib import Server
----> 2 Server("http://192.168.1.240/rpc").relay_get('al_nonexisting')

/home/jhenner/.local/lib/python2.7/site-packages/jsonrpclib/jsonrpc.pyc in __call__(self, *args, **kwargs)
    286                 '(according to JSON-RPC spec.)')
    287         if len(args) > 0:
--> 288             return self.__send(self.__name, args)
    289         else:
    290             return self.__send(self.__name, kwargs)

/home/jhenner/.local/lib/python2.7/site-packages/jsonrpclib/jsonrpc.pyc in _request(self, methodname, params, rpcid)
    236                         rpcid=rpcid, version=self.__version)
    237         response = self._run_request(request)
--> 238         check_for_errors(response)
    239         return response['result']
    240 

/home/jhenner/.local/lib/python2.7/site-packages/jsonrpclib/jsonrpc.pyc in check_for_errors(result)
    565         code = result['error']['code']
    566         message = result['error']['message']
--> 567         raise ProtocolError((code, message))
    568     return result
    569 

ProtocolError: (-32603, u'Internal Error')
**Expected behavior:**
Something like HTTP404 

**Actual behavior:** [What actually happened]
Internal Error

[forum]:https://forum.unipi.technology/category/4/official-evok-api
[google]:http://www.google.com/

tknot, 2018-06-15:
I will have a look at this, but the errors are issued by the JSON library we use, which means that it may be very difficult or almost impossible to change, at least without changing the library itself

kratochvil01, 2024-06-28:
[commit](https://github.com/UniPiTechnology/evok/pull/173/commits/7d06ccc9b27b9be4fb04e8640180117b87209ed6) 
I modified the return error in case of non-existent device to ['Invalid params' (code -32602)](https://www.jsonrpc.org/specification#error_object).

closed 2024-07-01 by kratochvil01