UnipiTechnology/evok #59: /version REST endpoint failure

alydersen, 2018-06-15:
* Running v 2.0.7 on Raspian Stretch. Fresh install

Calling the /version REST endpoint triggers an error:

From `evoke.py` running in console:

```
[E 180615 11:55:00 http1connection:54] Uncaught exception
    Traceback (most recent call last):
      File "/usr/local/lib/python2.7/dist-packages/tornado/http1connection.py", line 238, in _read_message
        delegate.finish()
      File "/usr/local/lib/python2.7/dist-packages/tornado/routing.py", line 251, in finish
        self.delegate.finish()
      File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 2097, in finish
        self.execute()
      File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 2117, in execute
        **self.handler_kwargs)
      File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 192, in __init__
        self.initialize(**kwargs)
      File "/opt/evok/evok.py", line 1247, in initialize
        with open('version.txt', 'r') as f:
    IOError: [Errno 2] No such file or directory: 'version.txt'
```
Expected behavior:
- Return the version number

Actual behavior:
- Throws an error and returns nothing

tknot, 2018-06-15:
At a quick glance this looks like an error in the installation script; I will properly test this ASAP

For a quick workaround it is possible to copy the version.txt file to the /opt/evok folder manually

tknot, 2018-09-21:
This should be fixed as of the current (v.2.0.7c) release

closed 2018-09-21 by tknot