UnipiTechnology/evok #61: Changing the default serial line settings to reflect defaults of the extensions

mkudlacek, 2018-07-09:
To simplify the usage of the EVOK together with UniPi extensions (namely xS50), it would be better to change the default serial line settings to 19200bps, 8N1, so the users would only need to set the correct address in the evok.conf

tknot, 2018-07-09:
This should be set to 19200/8N1 by default already, I'll have a look at this later.

tknot, 2018-11-05:
As of the latest release (2.0.7b):
The default is indeed 19200 8N1 (and address 15), both for the internal UART Modbus client as well as the extension settings. For the latter it is possible to change by overwriting the internal flash, or setting the dip-switch pins incorrectly.

tknot, 2018-11-05:
If there are persisting issues do no hesitate to open a new topic, but please do provide information about your configuration.

closed 2018-11-05 by tknot