UnipiTechnology/evok #63: Install script disables Wifi

kknuut, 2018-08-26:
### Prerequisites

* [X ] Are you running the latest Master or the latest release version?
* [ ] Did you try to reinstall EVOK?
* [ ] Did you perform a cursory search on the [forum] and [google]?

### Description

Install script asks if user wants to "install WiFi?". Warning message is shown to not install with custom network configuration.

If user has altered wlan configuration and is using WiFi to run script he/she quite naturally selects "no", in which case **ifconfig wlan0 down** command is executed! Causing drop connection if user was connected via wlan0 interface. 

### Steps to Reproduce

1. sudo bash install-evok.sh
2. Answer "no" to Install WiFi question

**Expected behavior:** script continues without touching wifi settings

**Actual behavior:**  ifconfig wlan0 down is executed  line  432

415-    # Setup wifi
416-    echo "############################################################################"
417:    echo "## !!! POTENTIALLY DANGEROUS: Do you wish to install WiFi AP support? !!! ##"
418-    echo "## !!! DO NOT USE WITH CUSTOM NETWORK CONFIGURATION                   !!! ##"
419-    echo "## !!! USE ONLY WITH PLAIN RASPBIAN STRETCH                           !!! ##"
420-    echo "############################################################################"
421-    echo ' '
422:    if ask "(Install WiFi?)"; then
423-            apt-get install -y hostapd dnsmasq iproute2
424-            systemctl disable hostapd dnsmasq
425-            systemctl stop hostapd dnsmasq
426-            sed -i -e 's/option domain-name/#option domain-name/' /etc/dhcp/dhcpd.conf
427-            sed -i -e 's/option domain-name-servers/#option domain-name-servers/' /etc/dhcp/dhcpd.conf
428-            sed -i -e 's/#authoritative/authoritative/' /etc/dhcp/dhcpd.conf
429-            sed -i -e 's/#net.ipv4.ip_forward=1/net.ipv4.ip_forward=1/' /etc/sysctl.conf
430-            sed -i -e 's/wifi_control_enabled = False/wifi_control_enabled = True/' etc/evok-neuron.conf
431-    else
432-            ifconfig wlan0 down
433-    fi

[forum]:https://forum.unipi.technology/category/4/official-evok-api
[google]:http://www.google.com/

tknot, 2018-08-27:
I can see how that would be an issue! I will fix this ASAP.

tknot, 2018-09-11:
Should be fixed as of v.2.0.7c

closed 2018-09-11 by tknot