UnipiTechnology/evok #64: Wrong kernel dependency in neutron-kernel

alydersen, 2018-09-14:
On following your instructions for installing EVOK, with the latest Raspbian stretch, I get:
```
root@raspberrypi:/home/pi# apt-get install neuron-kernel
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 neuron-kernel : Depends: raspberrypi-kernel (= 1.20180703-1) but 1.20180910-1 is to be installed
E: Unable to correct problems, you have held broken packages.
```

alydersen, 2018-09-14:
BTW: Same thing happened with unipi-modbus-tools.

tknot, 2018-09-21:
This should be fixed by a reboot after apt-get upgrade; the error message means that the raspberrypi kernel update is prepared but not yet executed, and so cannot be rolled back to the version required by Neuron. We have updated our documentation accordingly.

Unfortunately we have now moved to another method of installing our kernel driver (dkms) so there may be some other issues persisting, so I will keep this ticked open.