UnipiTechnology/evok #66: evok.log nekontrolovatelně narůstá

johnymachine, 2018-10-09:
https://github.com/UniPiTechnology/evok/blob/5a6dfcc5e668973fb9fa6177152d38ccb3067bfe/evok/neuron.py#L269

**evok.log nekontrolovatelně narůstá**

AttributeError: 'NoneType' object has no attribute 'do_scan'
2018-10-08 10:34:25,465 - evok - ERROR - 'NoneType' object has no attribute 'do_scan'
Traceback (most recent call last):
  File "/opt/evok/lib/python2.7/site-packages/evok/neuron.py", line 269, in scan_boards
    yield self.modbus_cache_map.do_scan()
AttributeError: 'NoneType' object has no attribute 'do_scan'
2018-10-08 10:34:25,566 - evok - ERROR - 'NoneType' object has no attribute 'do_scan'
Traceback (most recent call last):
  File "/opt/evok/lib/python2.7/site-packages/evok/neuron.py", line 269, in scan_boards
    yield self.modbus_cache_map.do_scan()
AttributeError: 'NoneType' object has no attribute 'do_scan'
2018-10-08 10:34:25,667 - evok - ERROR - 'NoneType' object has no attribute 'do_scan'
Traceback (most recent call last):
  File "/opt/evok/lib/python2.7/site-packages/evok/neuron.py", line 269, in scan_boards
    yield self.modbus_cache_map.do_scan()
AttributeError: 'NoneType' object has no attribute 'do_scan'
2018-10-08 10:34:25,768 - evok - ERROR - 'NoneType' object has no attribute 'do_scan'
Traceback (most recent call last):
  File "/opt/evok/lib/python2.7/site-packages/evok/neuron.py", line 269, in scan_boards
    yield self.modbus_cache_map.do_scan()
AttributeError: 'NoneType' object has no attribute 'do_scan'
2018-10-08 10:34:25,870 - evok - ERROR - 'NoneType' object has no attribute 'do_scan'
Traceback (most recent call last):
  File "/opt/evok/lib/python2.7/site-packages/evok/neuron.py", line 269, in scan_boards
    yield self.modbus_cache_map.do_scan()
AttributeError: 'NoneType' object has no attribute 'do_scan'
2018-10-08 10:34:25,972 - evok - ERROR - 'NoneType' object has no attribute 'do_scan'
Traceback (most recent call last):
  File "/opt/evok/lib/python2.7/site-packages/evok/neuron.py", line 269, in scan_boards
    yield self.modbus_cache_map.do_scan()

johnymachine, 2018-10-09:
I can provide remote acces to devices. I was on the phone with mr. Kudláček the other day ragarding unexpected resets.

Please DM me, so we can check the config.

Thanks.

tknot, 2018-12-12:
Opened again in order to account for the need for better error logging

tknot, 2018-12-12:
Logging behaviour has been improved as of the 6b7524957d3b16ed3d845c313b1165f367db3c10 commit

closed 2018-12-12 by tknot