UnipiTechnology/evok #71: EVOK WebSocket inconsistences

StevenBrs, 2019-02-18:
### Description

WebSocket answers are not consistent

### Steps to Reproduce

Setup : M203 with EVOK installed on raspbian + xS10 extension with hard-set serial settings

1a. Throw a message to the websocket to switch-on LED1 on M203 unit (or LED2, 3 and 4)
2a. Throw a message to the websocket to switch on Relay 2_01 on M203 unit
3a. Throw a message to the websocket to switch-on LED2, 3 or 4 on xS10 extension 
      (not LED1)

**Expected behavior:** [What you expected to happen]

1b. WS Answer : {"glob_dev_id": 1, "dev": "led", "circuit": "1_01", "value": 1}
2b. WS Answer : {"modes": ["Simple"], "relay_type": "physical", "value": 1, "glob_dev_id": 1, "dev": "relay", "circuit": "2_01", "pending": false, "mode": "Simple"}
3b. WS Answer : {"glob_dev_id": 1, "dev": "led", "circuit": "UART_1_02", "value": 1} 

**Actual behavior:** [What actually happened]

1b. All OK: Led is switched on, and the new status is published on the websocket
2b. Relay is switched on, but all statusses for all relays in that group are published, causing an overload of messages on the websocket, creating other problems (especially when switching multiple relays at the same time)
3b. LED is switched on, no new status-message is received from WebSocket, new state is not reflected in controlling software

Also see bullet-point 2 of this forum post : https://forum.unipi.technology/topic/766/evok-websocket-bugs-problems

[forum]:https://forum.unipi.technology/category/4/official-evok-api
[google]:http://www.google.com/

StevenBrs, 2019-02-18:
I forget about the "full" command in my expected behavior description.

"set" command: should just do what was asked (turn on led/relay/...)
"full" command should do "set" + publish new state on the websocket, preferable only for the mentioned output and not the whole group

wimstockman, 2019-09-27:
Found a small bug in the websocket "full" command. It returns a dictionary rather then a list containing a dictionary so if you are processing all the incoming message as a list you miss the "full" messages.
I will sent a pull request to fix it. and update my websocket examples.
kind regards

martytriska, 2019-10-02:
Yes, you are right. There is a bug in ws return value format for Unipi1.x. Since the fix is an api-changing modification, it will be introduced in the following major version release.

kratochvil01, 2024-07-18:
This problem is solved in Evok v3.