UnipiTechnology/evok #72: EVOK WebSocket Full-command not working

StevenBrs, 2019-02-18:
### Prerequisites

* [x] Are you running the latest Master or the latest release version?
* [x] Did you try to reinstall EVOK?
* [x] Did you perform a cursory search on the [forum] and [google]?

### Log files needed

# tail -n 5 /var/log/evok.log
2019-02-18 12:36:04,377 - evok - ERROR - Exc: full() takes exactly 1 argument (2 given)
2019-02-18 12:36:04,380 - evok - ERROR - Exc: full() takes exactly 1 argument (2 given)
2019-02-18 12:36:04,383 - evok - ERROR - Exc: full() takes exactly 1 argument (2 given)
2019-02-18 12:36:10,524 - evok - ERROR - Exc: full() takes exactly 1 argument (2 given)
2019-02-18 12:36:12,465 - evok - ERROR - Exc: full() takes exactly 1 argument (2 given)

### Description

EVOK WebSocket command "full" doesn't work.
"set" behaves sort of like "full" should do.

(Also see https://github.com/UniPiTechnology/evok/issues/71)

### Steps to Reproduce

Send websocket command : 
{"value": 1, "cmd": "set", "circuit": "1_04", "dev": "led"}

**Expected behavior:** [What you expected to happen]

Led turns on and new state is published to the WebSocket

**Actual behavior:** [What actually happened]

Nothing happens
EVOK error is logged in evok.log 

[forum]:https://forum.unipi.technology/category/4/official-evok-api
[google]:http://www.google.com/

tknot, 2019-02-19:
This looks like an issue with the LED device; I'll have a look at it in more detail shortly.

StevenBrs, 2019-02-19:
Hi tknot,
Happens also with relais

kratochvil01, 2024-04-10:
Hello, this issue is solved in beta Evok. You can try it [here](https://github.com/UniPiTechnology/evok/tree/dev-bookworm
).

kratochvil01, 2024-06-20:
Evok v3.0.1 is released.

closed 2024-06-20 by kratochvil01