UnipiTechnology/evok #73: EVOK UI watchdog settings not working

StevenBrs, 2019-02-18:
### Prerequisites

* [x] Are you running the latest Master or the latest release version?
* [x] Did you try to reinstall EVOK?
* [x] Did you perform a cursory search on the [forum] and [google]?

### Description

Cannot change the watchdog settings in the EVOK webinterface

### Steps to Reproduce

1. Go to webinterface (http://<ip/url>:88)
2. Choose configuration / board watchdogs / any WD (1.01, 2.01, UART 1 2.01)
3. Set command watchdog on "enabled", change timeout value to something else (eg 8000)
4. Press Submit
5. Reclick the WD and check values: still disabled and 5000ms
6. Do a REST call to get values --> value=0 and timeout=5000

**Expected behavior:** [What you expected to happen]

Settings are changed

**Actual behavior:** [What actually happened]

Settings are not changed

[forum]:https://forum.unipi.technology/category/4/official-evok-api
[google]:http://www.google.com/

kratochvil01, 2024-04-09:
Hello, this issue is solved in beta Evok. You can try it [here](https://github.com/UniPiTechnology/evok/tree/dev-bookworm
).

kratochvil01, 2024-06-20:
Evok v3.0.1 is released.

closed 2024-06-20 by kratochvil01