UnipiTechnology/evok #75: EVOK API how does relay_set_for_time work

aufi, 2019-03-30:
### Prerequisites

* [X] Are you running the latest Master or the latest release version?
* [X] Did you try to reinstall EVOK?
* [X] Did you perform a cursory search on the [forum] and [google]?

### Log files needed

Include the output of the following commands on your Neuron/UniPi device:
https://pastebin.com/jgvXDrQ7

(I use xS40 extension, which looks to work same (nearly fine) as main module now, so I'm not sure if log entries are related)

### Description

JSON RPC method ```relay_set_for_time``` doesn't work for me, it switch the output (which is OK), but does not return it back after timeout.

I compared my code to examples in forum and API docs and I'm either missing something or it doesn't work for me.

### Steps to Reproduce

1. Have latest released EVOK on raspbian on Neuron L203
2. Run following (ruby, but should be the same for other languages):
```
require "jsonrpc-client"
evok_rpc = JSONRPC::Client.new("http://evok_host")
evok_rpc.relay_set_for_time("2_11", 1, 10.0)
=> {"modes"=>["Simple"], "alias"=>"al_obyvak_stul_svetlo", "relay_type"=>"physical", "value"=>0, "glob_dev_id"=>1, "dev"=>"relay", "circuit"=>"2_11", "pending"=>true, "mode"=>"Simple"}
# and light turn on
```
**Expected behavior:**  Light (relay 2_11) will be switched off after 10 secs.

**Actual behavior:** Light (relay 2_11) stays on.

mkudlacek, 2019-04-15:
Hello @aufi, this is Martin from UniPi.

We were just walking through the current functionality and came across two problems: 
1) The procedure never worked as intended in the first place
2) The name of the procedure doesn't correspond with the functionality which you can read in the code

The intended functionality was to be able to invert the state of the relay after some time. We just had a discussion over what could be the usage of such function.  And since you are historically the first person trying to use this function, I would like to ask you about your intention so we can fix it properly. So, why do you need such function? What is your usecase?

The second problem is the naming, which suggests "set the relay <param_1> to value <param_2> for the <param_3> period of time". Basically this  would be a pulse generator which is being used in automation much more frequently than the function described above, and in our humble opinion, should be the correct behaviour.

Thank you for your reply in advance and have a nice day,
Martin

aufi, 2019-04-15:
Hi Martin, thanks for response!

My use-case is a jalousie/sunblinder switch. I planned switch engine on just for time until jalousie is pulled up/down and then turn it off. I have installed Somfy engines which have automatic switch off on fully open/closed state, but I can control jalousie from both button via direct switch and via Evok API and I need avoid situation when both directions (up and down) are switched on. Or at least decrease propability of this situation.

An alternative good solution would be to have an "exclusive flag" for group of relay outputs (implemented in _direct switch_) to ensure up and down switch is on in the same time. Hardware buttons have this "feature" by default, but when it becomes controlled by hardware buttons&API (e.g. too strong wind or blind all windows in house in the same time).

So far I worked with solution which proxies Evok events to MQTT [1] bus where several bots listens inclusing both that can switch off the "up" relay if "down" direction was requested (but I don't have it fully working ATM).

[1] https://github.com/aufi/evok-to-mqtt

matoushybl, 2026-10-08:
This feature has been obsolete for years, will be removed completely in 3.1.

miroslav-ondra, 2026-10-12:
Fixed in Evok 3.1 (branch `dev-3.1`), the pulse is not removed but done by a new sequencer (#244, commits 7789635, a85fb14):

- The RPC methods `relay_set_for_time` and `output_set_for_time` stay and work: the output is set to `value` and after `pulse_duration` seconds the opposite value is written. `output_get` returns `[value, pending]`.
- In the other APIs (REST, JSON, WebSocket, bulk) each output has its sequencer device, e.g. `POST /rest/sequencer/ro_2_11` with `value=1&pulse_duration=10`. It runs also a sequence of changes (`steps`) and can be cancelled (`cancel`), its state `running` shows a running pulse. The outputs themselves have no `pulse_duration` and `pending` anymore.
- A change of the output (e.g. by another request) cancels its pulse. When Evok stops during a pulse, the output is set to the end value of the pulse.

Documentation: `docs/apis/sequencer.md`.

The use case of the blinds (two relays which must not be on at the same time) is not covered by the sequencer, it would need another block.

aufi, 2026-10-12:
Awesome, thank you @miroslav-ondra!

closed 2026-11-02 by matoushybl