UnipiTechnology/evok #79: Installation fails, Raspberry Pi crashes at a certain point

hschretz, 2019-07-10:
I try to install EVOK on a Raspberry Pi 3B+, and I tried it as well on a model B Rev. 2, what takes same effekt.
The image ist 2019-06-20-raspbian-buster-lite.

Unipi v1.1

Connection is by WLAN (putty).

I2C is working.
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- 18 -- -- -- -- -- -- --
20: 20 -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: 50 -- -- -- -- -- -- 57 -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- 6f
70: -- -- -- -- -- -- -- --

Update and Upgrade done.

Installation like description for "Legacy installation process using a shell script (REQUIRED FOR UNIPI 1.1!)":
<<<<<
sudo su
wget https://github.com/UniPiTechnology/evok/archive/v.2.0.7h.zip
unzip v.2.0.7h.zip
cd evok-v.2.0.7h
bash install-evok.sh
>>>>>

Installation brakes every time after choice
      (Install WiFi?) [y/n] n

Outputs on the terminal:
<<<<<
## Installing EVOK for UniPi 1.x ##
###################################
############################
# Using device tree kernel #
############################
gcc -O3 -Wall -pthread -fpic -c -o pigpio.o pigpio.c
gcc -O3 -Wall -pthread -fpic -c -o command.o command.c
gcc -O3 -Wall -pthread -fpic -c -o pigpiod_if.o pigpiod_if.c
gcc -O3 -Wall -pthread -fpic -c -o pigpiod_if2.o pigpiod_if2.c
pigpio.c: In function ‘mbCreate’:
pigpio.c:2382:36: warning: implicit declaration of function ‘makedev’ [-Wimplicit-function-declaration]
    return mknod(dev, S_IFCHR|0600, makedev(MB_DEV_MAJOR, 0));
                                    ^~~~~~~
gcc -O3 -Wall -pthread   -c -o x_pigpio.o x_pigpio.c
gcc -O3 -Wall -pthread   -c -o x_pigpiod_if.o x_pigpiod_if.c
gcc -O3 -Wall -pthread   -c -o x_pigpiod_if2.o x_pigpiod_if2.c
gcc -O3 -Wall -pthread   -c -o pig2vcd.o pig2vcd.c
gcc -O3 -Wall -pthread   -c -o pigpiod.o pigpiod.c
gcc -O3 -Wall -pthread   -c -o pigs.o pigs.c
gcc -shared -o libpigpiod_if.so pigpiod_if.o command.o
gcc -shared -o libpigpiod_if2.so pigpiod_if2.o command.o
## Installing EVOK for UniPi 1.x ##
###################################
############################
# Using device tree kernel #
############################
gcc -O3 -Wall -pthread -fpic -c -o pigpio.o pigpio.c
gcc -O3 -Wall -pthread -fpic -c -o command.o command.c
gcc -O3 -Wall -pthread -fpic -c -o pigpiod_if.o pigpiod_if.c
gcc -O3 -Wall -pthread -fpic -c -o pigpiod_if2.o pigpiod_if2.c
pigpio.c: In function ‘mbCreate’:
pigpio.c:2382:36: warning: implicit declaration of function ‘makedev’ [-Wimplicit-function-declaration]
    return mknod(dev, S_IFCHR|0600, makedev(MB_DEV_MAJOR, 0));
                                    ^~~~~~~
gcc -O3 -Wall -pthread   -c -o x_pigpio.o x_pigpio.c
gcc -O3 -Wall -pthread   -c -o x_pigpiod_if.o x_pigpiod_if.c
gcc -O3 -Wall -pthread   -c -o x_pigpiod_if2.o x_pigpiod_if2.c
gcc -O3 -Wall -pthread   -c -o pig2vcd.o pig2vcd.c
gcc -O3 -Wall -pthread   -c -o pigpiod.o pigpiod.c
gcc -O3 -Wall -pthread   -c -o pigs.o pigs.c
strip --strip-unneeded libpigpiod_if.so
size     libpigpiod_if.so
   text    data     bss     dec     hex filename
  45051    3888   49244   98183   17f87 libpigpiod_if.so
gcc -o pig2vcd pig2vcd.o
strip --strip-unneeded libpigpiod_if2.so
size     libpigpiod_if2.so
   text    data     bss     dec     hex filename
  53584    3876    1844   59304    e7a8 libpigpiod_if2.so
gcc -o x_pigpiod_if x_pigpiod_if.o -L. -lpigpiod_if -pthread -lrt
gcc -o x_pigpiod_if2 x_pigpiod_if2.o -L. -lpigpiod_if2 -pthread -lrt
gcc -o pigs pigs.o command.o
gcc -shared -o libpigpio.so pigpio.o command.o
strip --strip-unneeded libpigpio.so
size     libpigpio.so
   text    data     bss     dec     hex filename
 196095    4796  593752  794643   c2013 libpigpio.so
gcc -o x_pigpio x_pigpio.o -L. -lpigpio -pthread -lrt
gcc -o pigpiod pigpiod.o -L. -lpigpio -pthread -lrt
/usr/bin/ld: ./libpigpio.so: undefined reference to `makedev'
collect2: error: ld returned 1 exit status
/usr/bin/ld: ./libpigpio.so: undefined reference to `makedev'
make: *** [Makefile:61: pigpiod] Fehler 1
make: *** Es wird auf noch nicht beendete Prozesse gewartet....
collect2: error: ld returned 1 exit status
make: *** [Makefile:52: x_pigpio] Fehler 1
gcc -o x_pigpio x_pigpio.o -L. -lpigpio -pthread -lrt
/usr/bin/ld: ./libpigpio.so: undefined reference to `makedev'
collect2: error: ld returned 1 exit status
make: *** [Makefile:52: x_pigpio] Fehler 1
############################################################################
## !!! POTENTIALLY DANGEROUS: Do you wish to install WiFi AP support? !!! ##
## !!! DO NOT USE WITH CUSTOM NETWORK CONFIGURATION                   !!! ##
## !!! USE ONLY WITH PLAIN RASPBIAN STRETCH                           !!! ##
############################################################################

(Install WiFi?) [y/n] n
>>>>>

Then raspberry pi crashes.

With best regards, hschretz

hschretz, 2019-07-10:
Supplement:

If choosing Install WiFi yes, installation finishes but doesn't work.

martytriska, 2019-07-11:
Hello, hschretz.
Unfortunately, the current version of Evok is not fully compatible and tested with the Debian (Raspbian) 10 Buster now. We are working on it. In the meantime, please use the previous version of Linux distro.

n9, 2019-09-13:
Any updates on Buster support?
Default Raspbian download offers Buster only: https://www.raspberrypi.org/downloads/raspbian/

matkapi, 2019-09-14:
In my case it  helped to install new version (71) of pigpio https://github.com/joan2937/pigpio/releases
After install and restart of RPI, pigpio daemon runs correctly

tomashora, 2019-09-16:
We have done so in the 2.1.5 testing release. If you wish to try, switch to the test repository (instead of main) in the /etc/apt/sources.d/unipi.list and perform apt update && apt install evok

martytriska, 2019-09-24:
To install Evok on Unipi 1.x and Unipi 1.x lite, follow steps from the readme:
<code>
sudo suwget https://github.com/UniPiTechnology/evok/archive/2.1.6.zip
unzip 2.1.6.zip
cd evok-2.1.6
bash install-evok.sh
</code>
Installation by the install script is no longer supported on Neuron and Axon series controllers. Use standard Debian package system apt.

martytriska, 2019-09-24:
To install Evok on Unipi 1.x and Unipi 1.x lite, follow steps from the Readme:

```
sudo su
wget https://github.com/UniPiTechnology/evok/archive/2.1.6.zip
unzip 2.1.6.zip
cd evok-2.1.6
bash install-evok.sh
```
Installation by the install-evok.sh script is no longer supported on Neuron and Axon series controllers. Use the standard debian package system instead.

closed 2024-04-09 by kratochvil01