UnipiTechnology/evok #81: Unipi 1.1 Installation Procedure not working for Raspberry Pi Error 404

rail3088, 2019-09-17:
### Prerequisites

* [ No] Are you running the latest Master or the latest release version?
* [ yes] Did you try to reinstall EVOK?
* [ yes] Did you perform a cursory search on the [forum] and [google]?

Installation procedure for raspberry pi for latest EVOK installation as per Github

sudo su
wget https://github.com/UniPiTechnology/evok/archive/v.2.0.7h.zip
unzip v.2.0.7h.zip
cd evok-v.2.0.7h
bash install-evok.sh

### Log files needed

root@ORMBLevelCrossingmon:/home/pi# sudo wget https://github.com/UniPiTechnology/evok/archive/v.2.0.7h.zip
--2019-09-17 11:57:16--  https://github.com/UniPiTechnology/evok/archive/v.2.0.7h.zip
Resolving github.com (github.com)... 13.236.229.21
Connecting to github.com (github.com)|13.236.229.21|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/UniPiTechnology/evok/zip/v.2.0.7h [following]
--2019-09-17 11:57:17--  https://codeload.github.com/UniPiTechnology/evok/zip/v.2.0.7h
Resolving codeload.github.com (codeload.github.com)... 3.105.64.153
Connecting to codeload.github.com (codeload.github.com)|3.105.64.153|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2019-09-17 11:57:17 ERROR 404: Not Found.

```
sudo su
cat /var/log/evok.access.log
ps -fax
cat /var/log/evok.log
```

### Description

Trying to install EVOK on raspberry pi to interact with UNIPI 1.1 on new build of Raspian.
When running command wget https://github.com/UniPiTechnology/evok/archive/v.2.0.7h.zip as per installation instructions receiving error 404 as above.
Looks like the archive folder (github.com/UniPiTechnology/evok/archive/) has been deleted.

Investigate what is the correct procedure to install EVOKE on a raspberry pi to drive unipi 1.1.

### Steps to Reproduce

1.     sudo su
2.     wget https://github.com/UniPiTechnology/evok/archive/v.2.0.7h.zip

**Expected behavior:** Install EVOK version 2.0.7 onto a raspberry pi

**Actual behavior:** received error 404 as per logs above.

[forum]:https://forum.unipi.technology/category/4/official-evok-api
[google]:http://www.google.com/

martytriska, 2019-09-23:
Of course, there was an issue in the document. It is fixed now. Please use the latest release:

```
sudo su
wget https://github.com/UniPiTechnology/evok/archive/2.1.6.zip
unzip 2.1.6.zip
cd evok-2.1.6
bash install-evok.sh
```

rail3088, 2020-07-23:
Thank You for your update confirmed working

closed 2020-07-23 by rail3088