UnipiTechnology/evok #82: Axon: Evok UI not showing input/output status

robertsLando, 2020-04-28:
Starting from this image: https://kb.unipi.technology/en:files:software:os-images:00-start?redirect=1#axon_opensource_os

I have installed evok using:

```
sudo su
apt-get update
apt-get upgrade
reboot

sudo su
apt-get install evok
systemctl enable evok
reboot
```

In install script I have selected to use a different port for evok and I used 8086.

Now when I open the UI it doesn't show atual status of the input/outputs, also logs in `/var/log/evok.log` are empty and service doesn't show any error.

martytriska, 2020-05-04:
Hello @robertsLando,

your problem can be initiated by two possible causes:

1) You MUST NOT disable the TCP server as Evok is using it internally.
2) The port used for EVOK API is defined in the configuration file /etc/evok.conf. No user-modifiable install script is used during installation from the Debian package on an Axon series PLC. Where have you changed the port?

robertsLando, 2020-05-04:
@martytriska If I enable unipitcp service it works but I cannot let it run as it uses modbus tcp port and I need that port. I would like to know if there is a way to stop just unipi modbus or run it on a separete port

martytriska, 2020-05-05:
@robertsLando OK, now it is a bit clearer. You can change the listening port used by unipitcp in file _/lib/systemd/system/unipitcp.service_ (and here you can also define the interface that the TCP server listen on - localhost is set by default, so you cannot connect to the server from outside the unit).

**BUT** on the Evok side, this port is a part of source code and cannot be changed by the user simply (as will be overwritten back with each upgrade). We consider adding this as a parameter to the /etc/evok.conf in future releases. 

TCP port 502 is the standard port mainly reserved for Modbus TCP and another use is not assumed usually.

Hoping this has been useful info for you!

robertsLando, 2020-05-05:
> TCP port 502 is the standard port mainly reserved for Modbus TCP and another use is not assumed usually.

I know that but I'm using a customized version of axon that is running another plc runtime (not mervis) but I need evok in order to map serial ports and to also have its web interface to see input/outputs status. So from what I have understand now this is not possible so I ask if there could be a feature for this: disable modbus tcp or move it to another port by exposing the porti in evok.conf file (defautl to 502)

martytriska, 2020-05-06:
Hello @robertsLando,

I understand you. As I have mentioned, the Modbus server has to be running together with Evok. Therefore, the change of the port is the only solution. You can add a line to the NEURON section of the /etc/evok.conf to specify the port. Example:

`
[NEURON_1]
global_id = 1
allow_register_access = False
scan_frequency = 10                          
scan_enabled = True   
modbus_port = 506
`

Port that the unipitcp listens on can be changed in:
`/lib/systemd/system/unipitcp.service`

Then restart both services (or reboot the PLC).

Please let me know if this was useful.

robertsLando, 2020-05-06:
@martytriska Yes it has been useful, thanks :)

closed 2020-05-06 by robertsLando