UnipiTechnology/evok #83: "Invalid device circuit number 1_01" fresh installation on Neuron S103

eLvErDe, 2020-07-16:
Hello,

I just unpacked a Neuron S103, flashed latest unipian image over the SD card and ran apt install evok.

Sadly, the I/O are not working:

```
root@S103-sn1611:~# /opt/evok/bin/python /opt/evok/lib/python2.7/site-packages/evok/evok.py
[I 200716 20:44:16 config:63] eprom: UniPi Neuron S103 version: 0.2 serial: 0x64b
[I 200716 20:44:16 evok:1738] Starting using config file /etc/evok.conf
[I 200716 20:44:16 config:87] YAML Definition loaded: xS51.yaml, type: 3, definition count 0
[I 200716 20:44:16 config:87] YAML Definition loaded: xS10.yaml, type: 3, definition count 1
[I 200716 20:44:17 config:87] YAML Definition loaded: xS50.yaml, type: 3, definition count 2
[I 200716 20:44:17 config:94] YAML Definition loaded: /etc/hw_definitions/BuiltIn/S103.yaml, type: UniPiBuiltIn
[I 200716 20:44:17 config:87] YAML Definition loaded: xS11.yaml, type: 3, definition count 3
[I 200716 20:44:17 config:87] YAML Definition loaded: xS40.yaml, type: 3, definition count 4
[I 200716 20:44:17 config:87] YAML Definition loaded: xS30.yaml, type: 3, definition count 5
[I 200716 20:44:17 config:87] YAML Definition loaded: CUSTOM_MODBUS_DEVICE.yaml, type: 3, definition count 6
[I 200716 20:44:17 config:87] YAML Definition loaded: evok-alias.yaml, type: 2, definition count 0
[I 200716 20:44:18 evok:1806] HTTP server listening on port: 8080
[I 200716 20:44:18 modbusclient_tornado:128] SPI client started
[I 200716 20:44:18 neuron:215] Reading SPI boards
[I 200716 20:44:18 neuron:230] No board on SPI 1
[I 200716 20:44:18 neuron:230] No board on SPI 2
[I 200716 20:44:18 owclient:444] Entering OWW loop with PID 1181
[I 200716 20:44:18 neuron:230] No board on SPI 3
[E 200716 20:44:18 config:384] Invalid device circuit number 1_01
    Traceback (most recent call last):
      File "/opt/evok/lib/python2.7/site-packages/evok/config.py", line 379, in add_aliases
        dev_obj = Devices.by_int(dev_pointer["dev_type"], dev_pointer["circuit"])
      File "/opt/evok/lib/python2.7/site-packages/evok/devices.py", line 67, in by_int
        raise Exception('Invalid device circuit number %s' % str(circuit))
    Exception: Invalid device circuit number 1_01
[E 200716 20:44:18 config:384] Invalid device circuit number 1_02
    Traceback (most recent call last):
      File "/opt/evok/lib/python2.7/site-packages/evok/config.py", line 379, in add_aliases
        dev_obj = Devices.by_int(dev_pointer["dev_type"], dev_pointer["circuit"])
      File "/opt/evok/lib/python2.7/site-packages/evok/devices.py", line 67, in by_int
        raise Exception('Invalid device circuit number %s' % str(circuit))
    Exception: Invalid device circuit number 1_02
```

Any idea what could be wrong ?

Thanks

martytriska, 2020-07-20:
Hello @eLvErDe ,

according to the log posted, communication with the extension I/O boards seems to be broken. It can be caused by various reasons e.g. kernel modules not loaded or internal Modbus TCP server is not working.

Please try to start with a fresh **opensource image** from unipi knowledge-base  [fresh latest opensource image from unipi knowledge-base](https://kb.unipi.technology/en:files:software:os-images:00-start#neuron_opensource_os) as unipian generation OS is obsolete.

If the problem still persists, please share the list of all running processes (_ps -ax_) and loaded kernel modules (_lsmod_).

eLvErDe, 2020-07-24:
Hello @martytriska 

Thanks for you response. Indeed it was related to unipi kernel module not being loaded but I have no clue how I ended up in such situation: kernel module index was "corrupted" and everything went back to normal after running `depmod`.
This does not make any sense to me, I'll watch kernel messages for SD card defects...

Regards, Adam.

closed 2020-07-24 by eLvErDe