UnipiTechnology/evok #84: Failure of update of Evok during Raspberry Pi upgrade

rail3088, 2020-07-23:
### Prerequisites

* [ n] Are you running the latest Master or the latest release version?
* [ Y] Did you try to reinstall EVOK?
* [Y ] Did you perform a cursory search on the [forum] and [google]?

### Log files needed

Include the output of the following commands on your Neuron/UniPi device:

```
sudo su
cat /var/log/evok.access.log
ps -fax
cat /var/log/evok.log
```

2020-07-23 14:18:06,916 - evok - ERROR - Exc: Invalid device circuit number 1_01
----
### Description

[Description of the bug or feature] Failure to complete Upgrade of Evok during normal apt-get upgrade proceedure.

### Steps to Reproduce

1. [First Step]                             apt-get  update, apt-get upgrade
2. [Second Step] Tried running wget -qO - https://repo.unipi.technology/debian/README.apt | bash
3. [and so on...]

**Expected behavior:** [What you expected to happen]
Upgrade complete
**Actual behavior:** [What actually happened]
 When Upgrade complete this error message,

Processing triggers for libvlc-bin:armhf (3.0.11-0+deb10u1+rpt1) ...
Errors were encountered while processing:
 unipi-kernel-modules-dkms
 unipi-modbus-tools

E: Sub-process /usr/bin/dpkg returned an error code (1)

I rebooted Unipi and tried upgrade again
pi@BASELevelcrossingmon:~ $ sudo apt-get upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages were automatically installed and are no longer required:
  alsa-base gstreamer0.10-alsa gstreamer0.10-plugins-base libdrm-etnaviv1 libgstreamer-plugins-base0.10-0 libgstreamer0.10-0 libllvm6.0 libmicrodns0 libva-wayland2
  libxfce4util-bin libxfce4util-common libxfce4util7 libxfconf-0-2 pimixer point-rpi rpi-eeprom-images xfconf
Use 'sudo apt autoremove' to remove them.
The following packages have been kept back:
  python-rpi.gpio python3-rpi.gpio raspberrypi-ui-mods scratch2
0 upgraded, 0 newly installed, 0 to remove and 4 not upgraded.
2 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] Y
Setting up unipi-kernel-modules-dkms (1.48) ...
Removing old unipi-1.48 DKMS files...

------------------------------
Deleting module version: 1.48
completely from the DKMS tree.
------------------------------
Done.
Loading new unipi-1.48 DKMS files...
It is likely that 5.4.51-v7+ belongs to a chroot's host
Building for 5.4.51+, 5.4.51-v7+, 5.4.51-v7l+ and 5.4.51-v8+
Building initial module for 5.4.51+
Error!  Build of unipi.ko failed for: 5.4.51+ (armv7l)
Consult the make.log in the build directory
/var/lib/dkms/unipi/1.48/build/ for more information.
dpkg: error processing package unipi-kernel-modules-dkms (--configure):
 installed unipi-kernel-modules-dkms package post-installation script subprocess returned error exit status 7
dpkg: dependency problems prevent configuration of unipi-modbus-tools:
 unipi-modbus-tools depends on unipi-kernel-modules (>= 1.42) | unipi-kernel-modules-dkms (>= 1.42) | neuron-kernel (>= 1.42) | axon-kernel (>= 1.13.20180719); however:
  Package unipi-kernel-modules is not installed.
  Package unipi-kernel-modules-dkms is not configured yet.
  Package neuron-kernel is not installed.
  Package axon-kernel is not installed.

dpkg: error processing package unipi-modbus-tools (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 unipi-kernel-modules-dkms
 unipi-modbus-tools
E: Sub-process /usr/bin/dpkg returned an error code (1)

Then I tried install procedure from Unipi site

pi@BASELevelcrossingmon:~ $ wget -qO - https://repo.unipi.technology/debian/README.apt | bash
Downloading and importing the Unipi repository signing key
OK

Downloading the APT definition for Unipi repository
deb https://repo.unipi.technology/debian buster main test

Setting preferences of APT to prefer production ready Unipi packages over test
Package: *
Pin: release o=unipi,c=main
Pin-Priority: 900

Updating the package list
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:3 https://deb.nodesource.com/node_12.x buster InRelease
Hit:4 http://packages.azlux.fr/debian buster InRelease
Get:5 https://repo.unipi.technology/debian buster InRelease [23.4 kB]
Get:6 https://repo.unipi.technology/debian buster/main all Packages [5,164 B]
Get:7 https://repo.unipi.technology/debian buster/main armhf Packages [17.2 kB]
Get:8 https://repo.unipi.technology/debian buster/test all Packages [15.6 kB]
Get:9 https://repo.unipi.technology/debian buster/test armhf Packages [46.5 kB]
Fetched 108 kB in 5s (23.5 kB/s)
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
unipi-firmware is already the newest version (5.50).
unipi-firmware set to manually installed.
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 unipi-kernel-modules : Depends: axon-kernel-image (= 1.20200601-1) but it is not installable or
                                 raspberrypi-kernel (= 1.20200601-1) but 1.20200717-1 is to be installed
E: Unable to correct problems, you have held broken packages.

It appears there is a issue between the what Evok is expecting in Kernel version and the latest downloaded and installed kernel.
 

[forum]:https://forum.unipi.technology/category/4/official-evok-api   Yes
[google]:http://www.google.com/  Yes

Raspian Version,
pi@BASELevelcrossingmon:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

raspberrypi-kernel-headers (1.20200717-1)

rail3088, 2020-07-23:
The Unipi is a M303 Neuron

kratochvil01, 2024-04-09:
Hello, you can try [beta evok](https://github.com/UniPiTechnology/evok/tree/dev-bookworm).

kratochvil01, 2024-06-24:
Evok v3 is released

closed 2024-06-24 by kratochvil01