UnipiTechnology/evok #85: Evok 2.3.0 crashes due to lackingmodule

paulfantom, 2020-08-10:
### Prerequisites

* [x] Are you running the latest Master or the latest release version?
* [x] Did you try to reinstall EVOK?
* [x] Did you perform a cursory search on the [forum] and [google]?

### Log files needed

Include the output of the following commands on your Neuron/UniPi device:

```
sudo su
cat /var/log/evok.access.log
ps -fax
cat /var/log/evok.log
```

Evok didn't start, so no logs were present.

### Description

[Description of the bug or feature]

### Steps to Reproduce

1. Flash raspbian buster onto new card
2. Start OS on RPi 4B+
3. Follow instructions from https://github.com/UniPiTechnology/evok#legacy-installation-process-using-a-shell-script-required-for-unipi-11 regarding installation of evok 2.3.0 on legacy system

**Expected behavior:** [What you expected to happen]
Evok starts

**Actual behavior:** [What actually happened]
Evok crashes with:
```
/usr/bin/python /opt/evok/evok.py
Traceback (most recent call last):
  File "/opt/evok/evok.py", line 39, in <module>
    import config
  File "/opt/evok/config.py", line 5, in <module>
    import neuron
  File "/opt/evok/neuron.py", line 7, in <module>
    from dali.bus import Bus
ImportError: No module named bus
```

[forum]:https://forum.unipi.technology/category/4/official-evok-api
[google]:http://www.google.com/

paulfantom, 2020-08-10:
Consider either pinning package `python-dali` to version 0.6 or upgrading code to be compatible with `python-dali>=0.7`

closed 2020-09-25 by martytriska

paulfantom, 2020-09-28:
It would be nice in the future to leave a comment on why the issue was closed and not forcing people to go through git commits to figure out by themselves if this was fixed or discarded.

martytriska, 2020-09-29:
Of course, you are righ. The issue has been fixed in 34395ee086c4b098913448200019d00eb18532ac .
Sorry for the inconvenience.