UnipiTechnology/evok #87: EVOK fails to install in image similar to Open Source OS

mhoejgaard, 2020-12-23:
### Prerequisites

* [Release ] Are you running the latest Master or the latest release version?
* [Yes ] Did you try to reinstall EVOK?
* [Yes ] Did you perform a cursory search on the [forum] and [google]?

### Log files needed

Include the output of the following commands on your Neuron/UniPi device:

```
sudo su
cat /var/log/evok.access.log
ps -fax
cat /var/log/evok.log
```

### Description

Cannot install EVOK in 2018-11-13-raspbian-stretch (which is what the Open Source OS is based on) using instructions in the README.

I am attempting an ancient image because Raspberry OS is hit by this issue: https://github.com/UniPiTechnology/evok/issues/84

Device: Unipi Neuron M193.
 
### Steps to Reproduce

1. Flash [](https://downloads.raspberrypi.org/raspbian/images/raspbian-2018-11-15/2018-11-13-raspbian-stretch.zip) to an SD card with BalenaEtcher.
2. Enable ssh by adding a file ssh to the boot partition
3. Boot the Unipi and follow the instructions here: (https://github.com/UniPiTechnology/evok#installation-process-on-neuron-family-controllers-with-fresh-rapsbian-image). 

**Expected behavior:** [What you expected to happen]
EVOK is installed.

**Actual behavior:** [What actually happened]
EVOK fails to install.

```
pi@raspberrypi:~ $ sudo su
root@raspberrypi:/home/pi# apt-get install nginx
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  fontconfig-config fonts-dejavu-core libfontconfig1 libgd3 libjbig0
  libnginx-mod-http-auth-pam libnginx-mod-http-dav-ext libnginx-mod-http-echo
  libnginx-mod-http-geoip libnginx-mod-http-image-filter
  libnginx-mod-http-subs-filter libnginx-mod-http-upstream-fair
  libnginx-mod-http-xslt-filter libnginx-mod-mail libnginx-mod-stream libtiff5
  libwebp6 libxpm4 libxslt1.1 nginx-common nginx-full
Suggested packages:
  libgd-tools fcgiwrap nginx-doc ssl-cert
The following NEW packages will be installed:
  fontconfig-config fonts-dejavu-core libfontconfig1 libgd3 libjbig0
  libnginx-mod-http-auth-pam libnginx-mod-http-dav-ext libnginx-mod-http-echo
  libnginx-mod-http-geoip libnginx-mod-http-image-filter
  libnginx-mod-http-subs-filter libnginx-mod-http-upstream-fair
  libnginx-mod-http-xslt-filter libnginx-mod-mail libnginx-mod-stream libtiff5
  libwebp6 libxpm4 libxslt1.1 nginx nginx-common nginx-full
0 upgraded, 22 newly installed, 0 to remove and 0 not upgraded.
Need to get 3,982 kB of archives.
After this operation, 8,423 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Get:1 http://mirrors.dotsrc.org/raspbian/raspbian stretch/main armhf libjbig0 armhf 2.1-3.1 [27.5 kB]
Get:2 http://mirrors.dotsrc.org/raspbian/raspbian stretch/main armhf fonts-dejavu-core all 2.37-1 [1,068 kB]
Get:3 http://mirrors.dotsrc.org/raspbian/raspbian stretch/main armhf fontconfig-config all 2.11.0-6.7 [271 kB]
Get:4 http://mirrors.dotsrc.org/raspbian/raspbian stretch/main armhf libfontconfig1 armhf 2.11.0-6.7 [313 kB]
Get:5 http://mirrors.dotsrc.org/raspbian/raspbian stretch/main armhf libtiff5 armhf 4.0.8-2+deb9u5 [220 kB]
Get:6 http://mirrors.dotsrc.org/raspbian/raspbian stretch/main armhf libwebp6 armhf 0.5.2-1 [202 kB]
Get:7 http://mirrors.dotsrc.org/raspbian/raspbian stretch/main armhf libxpm4 armhf 1:3.5.12-1 [43.6 kB]
[... 314 lines cut ...]
(If next boot fails, revert to initrd.img-4.19.66+.old-dkms image)
update-initramfs...(bad exit status: 1)
Warning: There was a problem remaking your initrd.  You must manually remake it
before booting into this kernel.

DKMS: uninstall completed.
Error! Problems with mkinitrd detected.  Automatically uninstalling this module.
DKMS: Install Failed (mkinitrd problems).  Module rolled back to built state.
dpkg: error processing package unipi-kernel-modules-dkms (--configure):
 subprocess installed post-installation script returned error exit status 7
dpkg: dependency problems prevent configuration of unipi-modbus-tools:
 unipi-modbus-tools depends on unipi-kernel-modules (>= 1.42) | unipi-kernel-modules-dkms (>= 1.42) | neuron-kernel (>= 1.42) | axon-kernel (>= 1.13.20180719); however:
  Package unipi-kernel-modules is not installed.
  Package unipi-kernel-modules-dkms is not configured yet.
  Package neuron-kernel is not installed.
  Package axon-kernel is not installed.

dpkg: error processing package unipi-modbus-tools (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 unipi-kernel-modules-dkms
 unipi-modbus-tools
E: Sub-process /usr/bin/dpkg returned an error code (1)

```

kratochvil01, 2024-04-09:
Hello, you can try [beta evok](https://github.com/UniPiTechnology/evok/tree/dev-bookworm).

kratochvil01, 2024-06-20:
Evok v3.0.1 is released.

closed 2024-06-20 by kratochvil01