UnipiTechnology/evok #90: Legacy install getting stuck on raspbian

hennjo, 2021-01-14:
### Prerequisites

* [No] Are you running the latest Master or the latest release version?
* [Yes ] Did you try to reinstall EVOK?
* [ Yes] Did you perform a cursory search on the [forum] and [google]?

### Log files needed

Include the output of the following commands on your Neuron/UniPi device:

```
sudo su
cat /var/log/evok.access.log
ps -fax
cat /var/log/evok.log
```
`pi@raspberrypi:~ $ sudo su
root@raspberrypi:/home/pi# cat /var/log/evok.access.log
cat: /var/log/evok.access.log: No such file or directory
root@raspberrypi:/home/pi# ps -fax
  PID TTY      STAT   TIME COMMAND
    2 ?        S      0:00 [kthreadd]
    3 ?        I<     0:00  \_ [rcu_gp]
    4 ?        I<     0:00  \_ [rcu_par_gp]
    5 ?        I      0:00  \_ [kworker/0:0-events]
    7 ?        I      0:00  \_ [kworker/u8:0-events_unbound]
    8 ?        I<     0:00  \_ [mm_percpu_wq]
    9 ?        S      0:00  \_ [ksoftirqd/0]
   10 ?        I      0:00  \_ [rcu_sched]
   11 ?        S      0:00  \_ [migration/0]
   12 ?        S      0:00  \_ [cpuhp/0]
   13 ?        S      0:00  \_ [cpuhp/1]
   14 ?        S      0:00  \_ [migration/1]
   15 ?        S      0:00  \_ [ksoftirqd/1]
   17 ?        I<     0:00  \_ [kworker/1:0H-kblockd]
   18 ?        S      0:00  \_ [cpuhp/2]
   19 ?        S      0:00  \_ [migration/2]
   20 ?        S      0:00  \_ [ksoftirqd/2]
   23 ?        S      0:00  \_ [cpuhp/3]
   24 ?        S      0:00  \_ [migration/3]
   25 ?        S      0:00  \_ [ksoftirqd/3]
   27 ?        I<     0:00  \_ [kworker/3:0H-kblockd]
   28 ?        S      0:00  \_ [kdevtmpfs]
   29 ?        I<     0:00  \_ [netns]
   30 ?        I      0:00  \_ [kworker/0:1-mm_percpu_wq]
   31 ?        I      0:00  \_ [kworker/1:1-events]
   32 ?        S      0:00  \_ [kauditd]
   33 ?        S      0:00  \_ [khungtaskd]
   34 ?        S      0:00  \_ [oom_reaper]
   35 ?        I<     0:00  \_ [writeback]
   36 ?        S      0:00  \_ [kcompactd0]
   54 ?        I<     0:00  \_ [kblockd]
   55 ?        I<     0:00  \_ [blkcg_punt_bio]
   56 ?        S      0:00  \_ [watchdogd]
   57 ?        I      0:00  \_ [kworker/2:1-events]
   59 ?        I<     0:00  \_ [rpciod]
   60 ?        I<     0:00  \_ [kworker/u9:0-hci0]
   61 ?        I<     0:00  \_ [xprtiod]
   62 ?        S      0:00  \_ [kswapd0]
   63 ?        I<     0:00  \_ [nfsiod]
   64 ?        I<     0:00  \_ [iscsi_eh]
   65 ?        I<     0:00  \_ [dwc_otg]
   66 ?        I<     0:00  \_ [DWC Notificatio]
   67 ?        I      0:00  \_ [kworker/1:2-events]
   68 ?        S<     0:01  \_ [vchiq-slot/0]
   69 ?        S<     0:00  \_ [vchiq-recy/0]
   70 ?        S<     0:00  \_ [vchiq-sync/0]
   71 ?        S      0:00  \_ [vchiq-keep/0]
   72 ?        S<     0:00  \_ [SMIO]
   73 ?        I      0:00  \_ [kworker/1:3-events]
[... 130 lines cut ...]

1. Install a fresh raspbian using a RPI3B+
2. Follow legacy installation instructions from README.md
3. get stuck :)

**Expected behavior:** I'd expect the installer to go through or a process that's doing some work.

**Actual behavior:** Output as above. If you change the command to sudo bash -x install-evok.sh you'll see that you get stuck on 

> + echo '####################################'
> ####################################
> + echo '## Using kernel newer than 3.18.5 ##'
> ## Using kernel newer than 3.18.5 ##
> + echo '####################################'
> ####################################
> + INCLUDE_LINE='include config_unipi.inc'
> + grep -q 'include config_unipi.inc'
> 

After changing that line to 
> if ! grep -f "$INCLUDE_LINE"; then
 the script ran succesfully.

[forum]:https://forum.unipi.technology/category/4/official-evok-api
[google]:http://www.google.com/

kratochvil01, 2024-04-09:
Hello, you can try [beta evok](https://github.com/UniPiTechnology/evok/tree/dev-bookworm).

kratochvil01, 2024-06-20:
Evok v3.0.1 is released.

closed 2024-06-20 by kratochvil01