UnipiTechnology/evok #93: unipi-modbus-tools 1.2.38 fills harddisk with logs

eLvErDe, 2021-03-24:
Hello,

Latest version is broken, I guess some debug logs are enabled but should not:

```
Mar 24 23:20:20 S103-sn27 unipi[1817]: FC: 3 Address is 1000, slave 0
Mar 24 23:20:20 S103-sn27 unipi[1817]: FC: 3 Address is 0, slave 0
Mar 24 23:20:20 S103-sn27 unipi[1817]: FC: 3 Address is 5, slave 0
Mar 24 23:20:20 S103-sn27 unipi[1817]: FC: 3 Address is 0, slave 0
Mar 24 23:20:20 S103-sn27 unipi[1817]: FC: 3 Address is 5, slave 0
Mar 24 23:20:20 S103-sn27 unipi[1817]: FC: 3 Address is 0, slave 0
Mar 24 23:20:20 S103-sn27 unipi[1817]: FC: 3 Address is 5, slave 0
Mar 24 23:20:21 S103-sn27 unipi[1817]: FC: 3 Address is 0, slave 0
Mar 24 23:20:21 S103-sn27 unipi[1817]: FC: 3 Address is 5, slave 0
Mar 24 23:20:21 S103-sn27 unipi[1817]: FC: 3 Address is 0, slave 0
Mar 24 23:20:21 S103-sn27 unipi[1817]: FC: 3 Address is 5, slave 0
Mar 24 23:20:21 S103-sn27 unipi[1817]: FC: 3 Address is 1000, slave 0
Mar 24 23:20:21 S103-sn27 unipi[1817]: FC: 3 Address is 0, slave 0
Mar 24 23:20:21 S103-sn27 unipi[1817]: FC: 3 Address is 5, slave 0
Mar 24 23:20:21 S103-sn27 unipi[1817]: FC: 3 Address is 0, slave 0
Mar 24 23:20:21 S103-sn27 unipi[1817]: FC: 3 Address is 5, slave 0
Mar 24 23:20:21 S103-sn27 unipi[1817]: FC: 3 Address is 0, slave 0
Mar 24 23:20:21 S103-sn27 unipi[1817]: FC: 3 Address is 5, slave 0
```

Rolling back to 1.2.36 suppress the logs.

Regards,

martytriska, 2021-03-26:
Hello @eLvErDe , 

yes, you are right, there is a bug. New version 1.2.40 will be released today.
Thank you for your report !!!

Have a nice day.

eLvErDe, 2021-03-26:
Great thank you :)

closed 2021-12-03 by eLvErDe