UnipiTechnology/evok #94: installing evok on debian bullseye - a bypass

rudyvan, 2021-11-29:
In debian buster, there was still support for pip2 and python-ow, but in the latest debian version, in bullseye, this support is dropped.
Therefore it is not possible to install evok on the latest release without some tricks.

I have been looking for a way around and below you find an ansible script that make this happen in an automated fashion.

When not versed in ansible, the things to do are:

- pull repo evok from git
- curl https://bootstrap.pypa.io/pip/2.7/get-pip.py --output get-pip.py
- check pip2 --version
- replace in install-evok.sh everywhere pip install with pip2 install
- install nginx before with sudo apt-get install nginx
- add line to /etc/apt/sources.list --> deb http://raspbian.raspberrypi.org/raspbian/ buster main contrib non-free rpi
- sudo apt-get update
- sudo ./install-evok.sh
- remove the added line in  /etc/apt/sources.list
- sudo apt-get update

goodluck!

``` ansible
- name: 1.installing evok - pull from git
  git:
    repo: https://github.com/UniPiTechnology/evok.git
    dest: "{{ home_dir }}/evok"
    update: yes
    force: True
  when: "'unipi' in myconfig and myconfig.unipi"
  tags:
    - sw
    - raspi

- name: 2.installing evok - place working /etc/evok.conf that was in use
  copy:
    src: "/home/rudyv/site_schoten/evok/{{ myconfig.hostname }}-evok.conf"
    dest: "/etc/evok.conf"
    backup: True
  become: True
  when: "'unipi' in myconfig and myconfig.unipi"
  tags:
    - sw
    - raspi

# pip for python2 is removed from latest pip versions, so add this now
# curl https://bootstrap.pypa.io/pip/2.7/get-pip.py --output get-pip.py
# check with: sudo pip2 --version
- name: 3.installing evok - fetch get-pip.py for pip2
  get_url:
    url: https://bootstrap.pypa.io/pip/2.7/get-pip.py
    dest: "{{ home_dir }}/evok/get-pip.py"
    force: yes
    mode: "0755"
  become: True
  when: "'unipi' in myconfig and myconfig.unipi"
  tags:
    - sw
    - raspi

- name: 3.installing evok - make get-pip.py for making pip2
  command: "python2 {{ home_dir }}/evok/get-pip.py"
  become: True
  when: "'unipi' in myconfig and myconfig.unipi"
  tags:
    - sw
    - raspi

- name: 4.installing evok - replace pip with pip2
  replace:
    path: "{{ home_dir }}/evok/install-evok.sh"
    regexp: '^()pip install (.*)'
    replace: '\1pip2 install \2'
  become: True
  when: "'unipi' in myconfig and myconfig.unipi"
  tags:
    - sw
    - raspi

- name: 5.installing evok - install nginx first
  apt:
    name: nginx
    update_cache: yes
  args:
    force_apt_get: True
  become: True
  when: "'unipi' in myconfig and myconfig.unipi"
  tags:
    - sw
    - raspi

# python-ow is dropped in higher versions than buster, is safe to add as merely bindings
- name: 6.installing evok - add raspian buster to apt sources
  lineinfile:
    dest: /etc/apt/sources.list
    regex: "^.*buster.*\n"
    line: "deb http://raspbian.raspberrypi.org/raspbian/ buster main contrib non-free rpi\n"
    state: present
  become: True
  when: "'unipi' in myconfig and myconfig.unipi"
  tags:
    - sw
    - raspi

- name: 7.installing evok - update apt
  apt:
    update_cache: yes
  args:
    force_apt_get: True
  become: True
  when: "'unipi' in myconfig and myconfig.unipi"
  tags:
    - sw
    - raspi

# run ./install-evok.sh with -d (unipi) or -l (unipi_light)
- name: 8.installing evok - run UniPi script
  ansible.builtin.expect:
    command: "./install-evok.sh -{{ myconfig.unipi_tpe }}"
    chdir: "{{ home_dir }}/evok"
    responses:
      (?i)Website Port to use: "8180"
      (?i)API Port to use: "8181"
      (?i)Install WiFi: "N"
      (?i)Do you want to overwrite your /etc/evok.conf file: "N"
      (?i)Is it OK to reboot now: "N"
    timeout: 180
  become: True
  when: "'unipi' in myconfig and myconfig.unipi"
  tags:
    - sw
    - raspi

- name: 9.installing evok - pause 5 seconds, service evok starting
  pause:
    seconds: 5
  when: "'unipi' in myconfig and myconfig.unipi"
  tags:
    - sw
    - raspi

- name: 10.installing evok - remove raspian buster from apt sources
  lineinfile:
    dest: /etc/apt/sources.list
    regex: "^.*buster.*\n"
    line: "deb http://raspbian.raspberrypi.org/raspbian/ buster main contrib non-free rpi\n"
    state: absent
  become: True
  when: "'unipi' in myconfig and myconfig.unipi"
  tags:
    - sw
    - raspi

- name: 11.installing evok - update apt again with buster removed
  apt:
    update_cache=yes
  args:
    force_apt_get: True
  become: True
  when: "'unipi' in myconfig and myconfig.unipi"
  tags:
    - sw
    - raspi

# this task stops running evok and restarts the evok service
- name: restart evok service, or start if not started
  ignore_errors: yes
  ansible.builtin.service:
    name: evok
    state: restarted
  become: True
  when: "'unipi' in myconfig and myconfig.unipi"
  tags:
    - evok_restart
    - raspi
    - sw

```

PEPITO82, 2022-11-05:
With the latest evok 2.4.20 release I could install it without any issues with latest bullseye 32 bit and 64 bit version.

martytriska, 2022-11-14:
Hello, as @PEPITO82 wrote, the latest releases are Debian 11 Bullseye compatible. We rework the install script and migrate 1Wire subsystem to onewire pip package instead of the python-ow Debian package that is no longer available.

kratochvil01, 2024-06-20:
Evok v3.0.1 is released.

closed 2024-06-20 by kratochvil01