UnipiTechnology/evok #95: unipi 1.1 analog out voltage is inversed with the software setting: 0 becomes pin 10V and 10V is pin 0V

rudyvan, 2021-12-11:
### Prerequisites

* [yes ] Are you running the latest Master or the latest release version?
* [ yes] Did you try to reinstall EVOK?
* [ yes] Did you perform a cursory search on the [forum] and [google]?

### Description

Measured on any unipi 1.1 version (tried several different hardware units in my possession, all the same result)

The analog output measured on the pins is inversed as per the settings via software:
  10V when via app set to zero and 0V when set to 10V

### Steps to Reproduce

install evok, select webserver port
via the unipi webserver change setting from 0 to 10V and measure the AO output with a multimeter, it is always the inverse

PEPITO82, 2022-11-05:
I have same issue with latest evok 2.4.20 and my unipi 1.1.

kratochvil01, 2024-04-09:
Hello, this issue is solved in beta Evok. You can try it [here](https://github.com/UniPiTechnology/evok/tree/dev-bookworm
).

kratochvil01, 2024-06-20:
Evok v3.0.1 is released.

closed 2024-06-20 by kratochvil01