UnipiTechnology/evok #98: Invalid syntax when starting evok.py

phillipsnick, 2022-05-27:
### Prerequisites

* [x] Are you running the latest Master or the latest release version? latest 2.4.12
* [x] Did you try to reinstall EVOK? Yes including fresh RPi image.
* [x] Did you perform a cursory search on the [forum] and [google]? 

### Log files needed

Include the output of the following commands on your Neuron/UniPi device:

cat /var/log/evok.access.log - not created
cat /var/log/evok.log - not created

### Description

Evok isn't starting

### Steps to Reproduce

**Expected behavior:** 

The software to run

**Actual behavior:** 

1. Attempt to start as service `sudo systemctl start evok`

Expect it to fail, check the status

```
● evok.service - Evok Modbus/Websocket/Rpc Server
     Loaded: loaded (/etc/systemd/system/evok.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Fri 2022-05-27 13:57:57 BST; 1min 6s ago
    Process: 497 ExecStart=/usr/bin/python /opt/evok/evok.py (code=exited, status=1/FAILURE)
   Main PID: 497 (code=exited, status=1/FAILURE)
        CPU: 89ms

May 27 13:57:57 raspberrypi systemd[1]: evok.service: Scheduled restart job, restart counter is at 5.
May 27 13:57:57 raspberrypi systemd[1]: Stopped Evok Modbus/Websocket/Rpc Server.
May 27 13:57:57 raspberrypi systemd[1]: evok.service: Start request repeated too quickly.
May 27 13:57:57 raspberrypi systemd[1]: evok.service: Failed with result 'exit-code'.
May 27 13:57:57 raspberrypi systemd[1]: Failed to start Evok Modbus/Websocket/Rpc Server.
May 27 13:58:54 raspberrypi systemd[1]: evok.service: Start request repeated too quickly.
May 27 13:58:54 raspberrypi systemd[1]: evok.service: Failed with result 'exit-code'.
May 27 13:58:54 raspberrypi systemd[1]: Failed to start Evok Modbus/Websocket/Rpc Server.
```

3. Run manually

```
sudo python /opt/evok/evok.py
  File "/opt/evok/evok.py", line 146
    except Exception,E:
                    ^
SyntaxError: invalid syntax
```

Expect syntax error

----

Python 3.9.2 installed

phillipsnick, 2022-09-14:
@mohagedorn yep. Use end of life Python 2

Looks like Unipi have abandoned this project as they have no plans to migrate to Python 3.

Kind of regret buying it to be honest! Pretty expensive paper weight at this point as I can't deploy something which no longer receives security updates into production.

mohagedorn, 2022-09-14:
Thanks for your quick reply.

kratochvil01, 2024-04-09:
Hello, you can try [beta evok](https://github.com/UniPiTechnology/evok/tree/dev-bookworm).

kratochvil01, 2024-06-20:
Evok v3.0.1 is released.

closed 2024-06-20 by kratochvil01